rockchip-linux/mpp #615: mpp_init失败返回mpp_nok

haozizhuimao, 2024-06-18:
在香橙派24系统上使用rkmpp编码时 mpp_init失败返回mpp_nok，怎么查看失败的原因

HermanChen, 2024-06-18:
先看 log

HermanChen, 2024-06-18:
串口上看 log 吧，linux 平台上 export mpp_syslog_perror=1 之后也能看得到

haozizhuimao, 2024-06-18:
我打开日志，报下面的错误，这个错误是不是vpu驱动没有安装，如果是我怎么确认vpu驱动是否安装了，谢谢

mpp[1215513]: mpi: mpp_create enter ctx 0xaaaaefe21188 mpi 0xaaaaefe21190
mpp[1215513]: mpp_info: mpp version: unknown mpp version for missing VCS info
mpp[1215513]: mpi: mpp_create leave ret 0 ctx 0xaaaaefe6eb60 mpi 0xffff928903b0
mpp[1215513]: mpi: mpi_control enter ctx 0xaaaaefe6eb60 cmd 200006 parm 0xfffff734ca94
mpp[1215513]: mpi: mpi_control leave ctx 0xaaaaefe6eb60 ret 0
mpp[1215513]: mpi: mpi_control enter ctx 0xaaaaefe6eb60 cmd 200007 parm 0xfffff734ca94
mpp[1215513]: mpi: mpi_control leave ctx 0xaaaaefe6eb60 ret 0
mpp[1215513]: mpi: mpp_init enter ctx 0xaaaaefe6eb60 type 1 coding 7
mpp[1215513]: vcodec_service: open vcodec_service /dev/mpp_service failed
mpp[1215513]: hal_h264e_vepu541: hal_h264e_vepu541_init mpp_dev_init failed. ret: -1
mpp[1215513]: mpp_enc_hal: mpp_enc_hal_init hal hal_h264e init failed ret -1
mpp[1215513]: mpp_enc_hal: mpp_enc_hal_init could not found coding type 7
mpp[1215513]: mpp_enc: mpp_enc_init_v2 could not init enc hal
mpp[1215513]: mpp: error found on mpp initialization
mpp[1215513]: mpi: mpp_init leave ctx 0xaaaaefe6eb60 ret -1

HermanChen, 2024-06-18:
/dev/mpp_service 设备有么

haozizhuimao, 2024-06-18:
/dev 目录下没有 mpp_service, 但有另外一块同样的板子也没有/dev/mpp_service 就可以编码

haozizhuimao, 2024-06-20:
问题已经到了，是瑞芯微的内核版本不一样导致，

closed 2024-06-20 by haozizhuimao