rockchip-linux/mpp #616: 请教mpp解码rtmp流decode_get_frame返回正常，但是frame为空的情况

dao027, 2024-06-19:
如题，mpp解码RTSP流的时候都正常，换成一个rtmp流的时候就一直是死循环，decode_get_frame(ctx, &frame)得到的frame一直是空

HermanChen, 2024-06-20:
把码流存下来先用 ffmpeg 验证看下是不是有问题

dao027, 2024-06-20:
> 把码流存下来先用 ffmpeg 验证看下是不是有问题

您好，
1、使用ffmpeg如下命令，保存的.h264流文件可以播放（30s的文件）
ffmpeg -i "rtmp://xxxx" -c:v copy -t 30 output.h264

使用mpi_dec_test -i output.h264 -t 7的demo测试上述保存的output.h264，基本正常，中途会报2个这样的错：
[image]
[image]

2、使用ffmpeg如下命令，硬解可以保存为yuv数据
ffmpeg -c:v h264_rkmpp -i "rtmp://xxxx" -vf "hwdownload,format=nv12" -t 3 output_nv12.yuv

保存后的yuv数据手动转mp4后可以正常播放

stz-source, 2024-07-17:
你好 请问你解决了吗

dao027, 2024-07-18:
> 你好 请问你解决了吗

还没有