rockchip-linux/mpp #617: mpp解码mjpeg再编码h264（低延迟）的大概流程？

circlefangzm, 2024-06-19:
看了一些例子和教程，几乎没有转码的，晕晕的。mjpeg按帧送给解码器（通过v4l2从摄像头取下来的1080P30fps流），宽高等参数已知，目标只有一个 **_低延迟_** ，还有几个疑问：
1. mjpeg转码成H264：应该用纯外部MppBuffer吗（纯外部buffer是不是有点相当于和显卡共享内存）？解码后帧还是在显存中是吗？是否可以直接给encoder进行编码？是否申请很多buffer（group方式）？
2（可选）. 如果解码后的图像，既要直接本地显示（HDMI全屏显示），又要编码成H264通过网络发出去（rtp什么的），一般怎么做？需要在显存中拷贝一份给RGA（我理解的零拷贝显示），另一份给encoder？
3（可选）. 上面需求1或需求2如果图像需要缩放或旋转，尤其是旋转，又该从哪里入手？
4. 低延迟是不是尽量提前配好参数，不要有info change？

之前试过ffmpeg，延迟好几秒，而且我还有一些摄像头的控制（v4l控制我能搞定），是不是只能自己写代码来降低延迟（转码延迟不知道是否可以控制在几十毫秒内）

谢谢

HermanChen, 2024-06-20:
参考 mpi_rc2_test 编码解码级联的处理

lxzlife, 2024-07-16:
请问楼主，你使用mjpeg硬件解码正常吗，我使用mpi_dec_test.c修改用来解码，解出来的yuv422sp有的能正常渲染，有的花屏，检查过mjepg源都能正常渲染 @circlefangzm

circlefangzm, 2024-07-16:
> 请问楼主，你使用mjpeg硬件解码正常吗，我使用mpi_dec_test.c修改用来解码，解出来的yuv422sp有的能正常渲染，有的花屏，检查过mjepg源都能正常渲染 @circlefangzm

能解码，但是drm显示我没弄成。其他drm显示正常。你用什么显示的？