rockchip-linux/mpp #618: 关于mpp的纯外部模式的问题

GuNail, 2024-06-20:
参考mpi_dec_utils 中的dec_buf_mgr_ 
[image]
启动后打印如下fd
[image]
但是解码后的frame获取fd打印
[image]
[image]

为啥解码后frame的fd和我外部分配的fd不一样，

HermanChen, 2024-06-20:
commit 之后会 dup fd，这样外面 fd 关闭不会影响里面，可以看下 import 之后的 fd，还有可以用 index 做匹配

GuNail, 2024-06-20:
> commit 之后会 dup fd，这样外面 fd 关闭不会影响里面，可以看下 import 之后的 fd，还有可以用 index 做匹配

好的，但是我没理解 “”import 之后的 fd ”  import  是指某个函数吗，并且我没有找到获取index的函数

HermanChen, 2024-06-21:
import 之后 mpp_buffer_get_index 可以得到 buffer 的 index

HermanChen, 2024-06-21:
都可以直接对接编码，参考 mpi_rc2_test.c

GuNail, 2024-06-26:
> 都可以直接对接编码，参考 mpi_rc2_test.c

感谢，目前实现了h264多路编解码，显示。我现在遇到个问题，在解码h265流，输出的画面异常，如下图：
[image]

log 如下
[image]

但是使用mpi_dec_test 解码没有问题，我仔细对比了下，h264和h265的配置在 mpp_init(ctx, MPP_CTX_DEC, type);
只是type 不一样

HermanChen, 2024-06-26:
265码流的vps/sps/pps头没有？还有看下输出的 MppFrame 的 stride 配置是否有不同

GuNail, 2024-06-26:
> 265码流的vps/sps/pps头没有？还有看下输出的 MppFrame 的 stride 配置是否有不同

2304X1080
[image]

GuNail, 2024-06-27:
> 265码流的vps/sps/pps头没有？还有看下输出的 MppFrame 的 stride 配置是否有不同

看了下文件 vps/sps/pps 是有的，我自己的demo 是通过ffmpeg 打开的h265文件put_packet 这样，但是ffmpeg 拉h265的rtsp流解码也这样
[image]

HermanChen, 2024-07-02:
什么芯片平台，是不是没有 265 编解码器

GuNail, 2024-07-02:
> 什么芯片平台，是不是没有 265 编解码器

rk3588，我用了mpi_dec_test 解H265 没问题，但是自己写的解h265就有问题，自己写的可以正常解h264，但是h264和h265的配置参数只有codeingtype 不一样，不过我的代码用了两个线程是一个put_packet 只放置码流，另一个get_frame

HermanChen, 2024-07-03:
双线程的话参考 mpi_dec_mt_test 看看，正常应该是一致的，是不是分帧什么的有区别

GuNail, 2024-07-03:
> 双线程的话参考 mpi_dec_mt_test 看看，正常应该是一致的，是不是分帧什么的有区别

好的，我看看，不过我的代码也开了分帧