rockchip-linux/mpp #624: mpp编码时，rk_vcodec报错

ccDraw, 2024-06-24:
报错信息
```shell
Jun 24 17:02:05 cdhf kernel: [28754.791654] rk_vcodec: mpp_task_timeout_work:567: task 18078:311:700790 processing time out!
Jun 24 17:02:05 cdhf kernel: [28754.791745] rk_vcodec: task 700790 dump timing at 28750394274 us:
Jun 24 17:02:05 cdhf kernel: [28754.791757] rk_vcodec: timing: create         : 0 us
Jun 24 17:02:05 cdhf kernel: [28754.791768] rk_vcodec: timing: create end     : invalid
Jun 24 17:02:05 cdhf kernel: [28754.791778] rk_vcodec: timing: pending        : invalid
Jun 24 17:02:05 cdhf kernel: [28754.791788] rk_vcodec: timing: run            : invalid
Jun 24 17:02:05 cdhf kernel: [28754.791797] rk_vcodec: timing: timeout start  : invalid
Jun 24 17:02:05 cdhf kernel: [28754.791806] rk_vcodec: timing: run end        : invalid
Jun 24 17:02:05 cdhf kernel: [28754.791816] rk_vcodec: timing: irq            : invalid
Jun 24 17:02:05 cdhf kernel: [28754.791825] rk_vcodec: timing: timeout cancel : invalid
Jun 24 17:02:05 cdhf kernel: [28754.791833] rk_vcodec: timing: isr            : invalid
Jun 24 17:02:05 cdhf kernel: [28754.791843] rk_vcodec: timing: finish         : invalid
Jun 24 17:02:05 cdhf kernel: [28754.791859] mpp_rkvenc2 fdbd0000.rkvenc-core: resetting...
Jun 24 17:02:05 cdhf kernel: [28754.791910] mpp_rkvenc2 fdbd0000.rkvenc-core: reset done
Jun 24 17:02:05 cdhf kernel: [28754.793395] mpp_rkvenc2 fdbd0000.rkvenc-core: can not activate fdbd0000.rkvenc-core -> fdbd0000.rkvenc-core
```
我在RK3588上使用mpp编码时报如上错误，编码格式为MPP_FMT_YUV400，给的数据区域的格式时rga的RK_FORMAT_YCbCr_400（RK_FORMAT_Y4也不对）有遇到过类似的情况吗。但我rga的格式为RK_FORMAT_YUYV_420这些的时候，编码参数给RK_FORMAT_YUYV_422这些就没有问题，可以正常编出
rga版本为1.9.0

HermanChen, 2024-06-24:
mpp 有更新 develop 分支么？

ccDraw, 2024-06-25:
> mpp 有更新 develop 分支么？

没有更过。今天早上发现问题了，是我给编码的数据大小不对，给的数据长度超过了实际的数据长度导致的。

closed 2024-06-25 by ccDraw