rockchip-linux/mpp #625: mpp version: unknown mpp version for missing VCS info

czyt, 2024-06-25:
hello,recently this issue occured in my project,but I have tried many ways,but not working.so I make this issue for help.thanks.
```
mpp[2545]: mpi: mpp_create enter ctx 0xaaab1e749bc8 mpi 0xaaab1e749bd0
mpp[2545]: mpp_info: mpp version: unknown mpp version for missing VCS info
mpp[2545]: mpi: mpp_create leave ret 0 ctx 0xaaab1e748840 mpi 0xffffaf5303b0
mpp[2545]: mpi: mpi_control enter ctx 0xaaab1e748840 cmd 200006 parm 0xffffd90025e4
mpp[2545]: mpi: mpi_control leave ctx 0xaaab1e748840 ret 0
mpp[2545]: mpi: mpi_control enter ctx 0xaaab1e748840 cmd 200007 parm 0xffffd90025e4
mpp[2545]: mpi: mpi_control leave ctx 0xaaab1e748840 ret 0
mpp[2545]: mpi: mpp_init enter ctx 0xaaab1e748840 type 1 coding 7
mpp[2545]: vcodec_service: open vcodec_service /dev/mpp_service failed
mpp[2545]: hal_h264e_vepu541: hal_h264e_vepu541_init mpp_dev_init failed. ret: -1
mpp[2545]: mpp_enc_hal: mpp_enc_hal_init hal hal_h264e init failed ret -1
mpp[2545]: mpp_enc_hal: mpp_enc_hal_init could not found coding type 7
mpp[2545]: mpp_enc: mpp_enc_init_v2 could not init enc hal
mpp[2545]: mpp: error found on mpp initialization
mpp[2545]: mpi: mpp_init leave ctx 0xaaab1e748840 ret -1
mpp_init-1
EncodeManage::init FFMPEGEncodeVideo
```

czyt, 2024-06-25:
the issue ware caused by permit.

closed 2024-06-25 by czyt

neroes, 2024-12-03:
i'm seeing a similar issue, what permits did you add?