rockchip-linux/mpp #627: the plan for implementation ISP APIs of MPI

plumnew, 2024-06-26:
I would like to konw when this parts implememted.
now i am seeking a solution to feed the isp processer raw images directly then get the results images back and during the same time, we could set the isp parameters.
from the wiki description https://opensource.rock-chips.com/wiki_Rockchip-isp1, it seems not possible change the pipeline to get the image from memory directly.
[image]
many thanks~

HermanChen, 2024-06-26:
ISP is not in mpp libary.

plumnew, 2024-06-26:
https://github.com/rockchip-linux/mpp/blob/develop/mpp/mpi.cpp
static MPP_RET mpi_isp(MppCtx ctx, MppFrame dst, MppFrame src);
static MPP_RET mpi_isp_put_frame(MppCtx ctx, MppFrame frame);
static MPP_RET mpi_isp_get_frame(MppCtx ctx, MppFrame *frame);

in this file, there are three reserved APIs for ISP functions. Are that deprecated? if it would not support in mpp, where could i directly access these APIs?  Are there suggestions for this use case? thanks~

HermanChen, 2024-07-02:
These api is not implemented and reserved for future video pre/post-process.

closed 2024-07-02 by plumnew