rockchip-linux/mpp #628: ioctl hangs on first try to deque frame from camera source

belveder79, 2024-07-03:
As we are working on the OV9281 and OV4689 on NanoPC-T6 on the 5.10.160 kernel, we found an issue in the [camera_source_ioctl](https://github.com/rockchip-linux/mpp/blob/239e15eaebf98f68c0093b00ff8108d23b0147e7/utils/camera_source.c#L91
) function.

The first time the camera is opened, the ioctl call in the while() loop hangs forever, it never returns, which makes everything hang. I added a few more logs to identify this issue:
- The "Get Frame" message is [here](https://github.com/rockchip-linux/mpp/blob/239e15eaebf98f68c0093b00ff8108d23b0147e7/utils/camera_source.c#L314) right before entering the camera_source_get_frame function.
- The "Deque" message is [here](https://github.com/rockchip-linux/mpp/blob/239e15eaebf98f68c0093b00ff8108d23b0147e7/utils/camera_source.c#L383) right before entering the camera_source_ioctl function.

```
Jul 03 11:51:37 qurve0test mpp[753]: camera_source: get width 1280 height 720
Jul 03 11:51:37 qurve0test mpp[753]: camera_source: get dma buf(0)-fd: 13
Jul 03 11:51:37 qurve0test mpp[753]: camera_source: get dma buf(1)-fd: 14
Jul 03 11:51:37 qurve0test mpp[753]: camera_source: get dma buf(2)-fd: 15
Jul 03 11:51:37 qurve0test mpp[753]: camera_source: get dma buf(3)-fd: 16
Jul 03 11:51:37 qurve0test mpp[753]: camera_source: Starting to set up buffers...
Jul 03 11:51:37 qurve0test mpp[753]: camera_source: Starting to capture...
Jul 03 11:51:37 qurve0test env[643]: DBG: /dev/media1: wait stream start event success ...
Jul 03 11:51:37 qurve0test env[643]: DBG: /dev/media1: wait stream stop event...
Jul 03 11:51:37 qurve0test kernel: rkisp_hw fdcb0000.rkisp: set isp clk = 396000000Hz
Jul 03 11:51:37 qurve0test kernel: rkcif-mipi-lvds2: stream[0] start streaming
Jul 03 11:51:37 qurve0test kernel: rkcif-mipi-lvds2: Allocate dummy buffer, size: 0x003f5000
Jul 03 11:51:37 qurve0test kernel: rockchip-mipi-csi2 fdd30000.mipi2-csi2: stream on, src_sd: 00000000fea294cc, sd_name:rockchip-csi2-dphy0
Jul 03 11:51:37 qurve0test kernel: rockchip-mipi-csi2 fdd30000.mipi2-csi2: stream ON
Jul 03 11:51:37 qurve0test kernel: rockchip-csi2-dphy0: dphy0, data_rate_mbps 1000
Jul 03 11:51:37 qurve0test kernel: rockchip-csi2-dphy csi2-dphy0: csi2_dphy_s_stream stream on:1, dphy0
Jul 03 11:51:37 qurve0test kernel: rockchip-csi2-dphy csi2-dphy0: csi2_dphy_s_stream stream on:1, dphy0
Jul 03 11:51:37 qurve0test mpp[753]: camera_source: Skipping some 4 frames...
Jul 03 11:51:37 qurve0test mpp[753]: camera_source: Get Frame 0...
Jul 03 11:51:37 qurve0test mpp[753]: camera_source: Dequeue buffer...
```

@HermanChen any ideas how to fix this? thx...

HermanChen, 2024-07-03:
It is not a part of mpp. So it is better to create an issue on rockchip redmine system.

belveder79, 2024-07-03:
> It is not a part of mpp. So it is better to create an issue on rockchip redmine system.
thx...

closed 2024-07-03 by belveder79