rockchip-linux/mpp #629: h264解码有时候会卡在decode_get_frame

cai6400, 2024-07-03:
使用h264格式打开摄像头，预览时有时会出现获取第一帧解码后的数据卡在decode_get_frame的问题。

cai6400, 2024-07-03:
```
        MppPollType timeout = MPP_POLL_BLOCK;
        MppParam param = &timeout;

        ret = cam_dev->dec_data->mpi->control(cam_dev->dec_data->ctx, MPP_SET_OUTPUT_TIMEOUT, param);
        if (ret) {
            USB_CAMERA_LOGD("Failed to set output timeout %d ret %d\n", timeout, ret);
            return -1;
        }
```
设置的输出模式如上

cai6400, 2024-07-03:
现在使用的是纯外部分配内存模式

HermanChen, 2024-07-17:
可以加一个 base:fast_out 的配置
从码流语法看，不一定能输入一帧就输出一帧

cai6400, 2024-07-17:
这是来自QQ邮箱的假期自动回复邮件。你好，我无法亲自回复你的邮件。我将在以后，尽快给你回复。

cai6400, 2024-07-17:
我还有一个问题，rk3588使用hwjpeg解码mjpeg时，4K摄像头码率较高时预览一段时间后会卡在解码过程，导致预览画面消失，不能恢复，如果使用低码率的4K摄像头或者1080P分辨率不会出现卡在解码的问题。
加打印发现是卡在ret = mMpi->poll(mMppCtx, MPP_PORT_OUTPUT, MPP_POLL_BLOCK);

cai6400, 2024-07-19:
mjpeg解码卡住看起来和一帧数据太大有关