rockchip-linux/mpp #630: rk3568上mpp解码后由drm显示的问题

zeonman, 2024-07-11:
通过MPP_DEC_SET_OUTPUT_FORMAT设置为MPP_FMT_YUV420SP | MPP_FRAME_FBC_AFBC_V2后,
modifier = DRM_FORMAT_MOD_ARM_AFBC(AFBC_FORMAT_MOD_BLOCK_SIZE_16x16 | AFBC_FORMAT_MOD_SPLIT | AFBC_FORMAT_MOD_SPARSE);
alg_width = Align(width, 4);
alg_height = Align(width, 4);
uint32_t strides[4] = { alg_width, 0, 0, 0 };
uint32_t offsets[4] = { 0 };
uint64_t modifiers[4] {modifier, 0,  0, 0 };
drmModeAddFB2WithModifiers(fd, alg_width, alg_height, DRM_FORMAT_NV12,handles, strides, offsets, modifiers,&fb_id, DRM_MODE_FB_MODIFIERS);
drmModeAddFB2WithModifiers返回-22,请问是什么原因呢

HermanChen, 2024-07-11:
这个是 drm 相关的问题，需要看对应内核的 drm 驱动了，不属于 mpp 内部的问题，不好分析……

HermanChen, 2024-07-11:
/* HACK: Fake hstride for Rockchip DRM driver
     *
     * When importing AFBC dma-bufs, the Rockchip DRM driver would calculate
     * the pixel stride from pitch. But the pitch aligning algorithms are
     * different between MPP and the driver:
     *
     * MPP uses round_up_64(round_up_64(width) * bpp / 8)
     * Rockchip DRM driver expects (pixel_stride * bpp / 8)
     *
     * We need the driver to use the same pixel stride as MPP, so we
     * re-calculate a fake hstride from the pixel stride (i.e. the 64-aligned
     * fake width) here.
     *
     * NOTE: The hstride is not used by others for now.
     */
需要有个 trick 让 drm 能被底层正常计算到

closed 2024-07-17 by HermanChen