rockchip-linux/mpp #631: rk3588 jpeg连续解码摄像头图像有花屏

rxs20032003, 2024-07-11:
修改mpp的mpi_dec_test例子，添加camera_source.c的接口采集usb 摄像头的mjpeg数据，解码后查看保存的yuv文件，发现第四张以后，图片有花屏的现象，出现很多横条。但是连续接一张jpeg图片接很多次不会出现该现象。
修改方式：在dec_advanced中添加如下代码，摄像头初始化代码在循环外面。
    cam_frm_idx = camera_source_get_frame(data->cam_ctx);
    mpp_assert(cam_frm_idx >= 0);

    cam_buf = camera_frame_to_buf(data->cam_ctx, cam_frm_idx);
    mpp_assert(cam_buf);
    mpp_packet_init_with_buffer(&packet, cam_buf);
[image]

HermanChen, 2024-07-12:
这看起来像是编码和采集有对同一个 buffer 进行访问？
码流解码时候有报错么？

rxs20032003, 2024-07-12:
没有报错，每张都解出来了，就是后面的全部都显示异常。采集返回的MppBuffer对输入的MppPacket进行了初始化，好像没有用同一个buffer。

HermanChen, 2024-07-12:
看起来是 buffer 管理有问题，采集端还在采集的 buffer 给编码器用了
打印下采集的 buffer 使用和编码的 buffer 使用看看

rxs20032003, 2024-07-12:
应该是解码器吧，我采集的用的是camera_source.c的接口，解码用的是mpi_dec_test里面例子，基本都没有改动，就是输入的packet用camera_source返回的buffer。我保存了buffer里面的jpeg数据，画面都是完整的

rxs20032003, 2024-07-12:
我发现jpeg正常的图片取决于camera_source_init初始化传递的bufcnt，这个参数没有影响jpeg的获取，影响到了jpeg的解码。前bufcnt张的yuv图片没有问题，后面就开始乱了，感觉像是有乱的数据一样。

HermanChen, 2024-07-12:
cache 问题，在给 mpp 去解码之前，调用下 mpp_buffer_sync_begin

rxs20032003, 2024-07-12:
我加上试了一下 还是不行，只要是在解码之前，放哪里都是一样的吧，解码之后调了end.但是没啥改善~~~

HermanChen, 2024-07-15:
不一样，要在码流送解码之前调用，把数据写入 ddr 让硬件真正访问到

lxzlife, 2024-07-16:
同样的问题，采集了100张mjpeg图像送给mpp解码，部分渲染正常，部分渲染不正常

lxzlife, 2024-07-17:
在enqueue(MPP_PORT_INPUT)之前调用了mpp_buffer_sync_begin，效果好了很多，但还是有小部分花屏 @HermanChen 
如图：
[image]

lxzlife, 2024-07-17:
[image]

HermanChen, 2024-07-17:
上面彩色的部分应该是宽高 stride 配置有问题，比如 1080 要变成 1088 这样对齐的宽高

lxzlife, 2024-07-17:
@HermanChen 
[image]

HermanChen, 2024-07-17:
色度地址是从亮度地址往后 h_stride * v_stride 的地址开始，不是从 w * h 的位置开始

lxzlife, 2024-07-17:
@HermanChen 没问题了，感谢解答，让我明白了这个stride用在这；万分感谢

closed 2024-07-17 by HermanChen

rxs20032003, 2024-07-17:
我的还不行~~ 我暂时用软解码实现了，有空帮忙看看啥问题，用的就是demo的解码例子，加了几行采集的代码。

	

    cam_frm_idx = camera_source_get_frame(data->cam_ctx);
    mpp_assert(cam_frm_idx >= 0);

    cam_buf = camera_frame_to_buf(data->cam_ctx, cam_frm_idx);
	
    mpp_assert(cam_buf);
	
    mpp_packet_init_with_buffer(&packet, cam_buf);

	
    Limelog("task IN\n");
    ret = mpi->poll(ctx, MPP_PORT_INPUT, MPP_POLL_BLOCK);
    if (ret) {
        printf("%p mpp input poll failed\n", ctx);
        return ret;
    }

    ret = mpi->dequeue(ctx, MPP_PORT_INPUT, &task);  /* input queue */
    if (ret) {
        printf("%p mpp task input dequeue failed\n", ctx);
        return ret;
    }

    mpp_assert(task);

    mpp_task_meta_set_packet(task, KEY_INPUT_PACKET, packet);
    mpp_task_meta_set_frame (task, KEY_OUTPUT_FRAME,  frame);
    mpp_buffer_sync_begin(cam_buf);
    ret = mpi->enqueue(ctx, MPP_PORT_INPUT, task);  /* input queue */
    if (ret) {
        printf("%p mpp task input enqueue failed\n", ctx);
        return ret;
    }

    if (!data->first_pkt)
        data->first_pkt = mpp_time();

    /* poll and wait here */
    ret = mpi->poll(ctx, MPP_PORT_OUTPUT, MPP_POLL_BLOCK);
    if (ret) {
        printf("%p mpp output poll failed\n", ctx);
        return ret;
    }
    Limelog("task OUT\n");

    ret = mpi->dequeue(ctx, MPP_PORT_OUTPUT, &task); /* output queue */
    if (ret) {
        printf("%p mpp task output dequeue failed\n", ctx);
        return ret;
    }

    mpp_assert(task);

    if (task) {
        MppFrame frame_out = NULL;

        mpp_task_meta_get_frame(task, KEY_OUTPUT_FRAME, &frame_out);

        if (frame) {
            if (!data->first_frm)
                data->first_frm = mpp_time();

            Limelog("dump_mpp_frame_to_file IN\n");
          /* write frame to file here */
            {
	            if (data->fp_output)
	                dump_mpp_frame_to_file(frame, data->fp_output);
			}
			RK_U32 error,discard;
			error = mpp_frame_get_errinfo(frame_out); 
	        if (error) {
	            discard = mpp_frame_get_discard(frame_out);
	            printf("Received a errinfo frame. error = %d, discard = %d\n", error, discard);
	        }
            if (data->fp_verify) {
                calc_frm_crc(frame, checkcrc);
                write_frm_crc(data->fp_verify, checkcrc);
            }
            Limelog("dump_mpp_frame_to_file OUT\n");

            Limelog("%p decoded frame %d\n", ctx, data->frame_count);
            data->frame_count++;

            if (mpp_frame_get_eos(frame_out)) {
                printf(quiet, "%p found eos frame\n", ctx);
            }
            fps_calc_inc(cmd->fps);
        }

        if (data->frame_num > 0) {
            if (data->frame_count >= data->frame_num)
                data->loop_end = 1;
        } else if (data->frame_num == 0) {
            if (slot->eos)
                data->loop_end = 1;
        }

        /* output queue */
        ret = mpi->enqueue(ctx, MPP_PORT_OUTPUT, task);
        if (ret)
            printf("%p mpp task output enqueue failed\n", ctx);
    }
        mpp_packet_deinit(&packet);
    	camera_source_put_frame(data->cam_ctx, cam_frm_idx);
	mpp_buffer_sync_end(cam_buf);

tubzby, 2025-09-05:
我也碰到一样的问题： https://github.com/rockchip-linux/mpp/issues/898

如camera_source_init 中buf_cnt == 4，解析出来前面4个yuv 是正常，后边就是花的。

我再 enqueue(MPP_PORT_INPUT) 之前执行 mpp_buffer_sync_begin(pkt_buf)没有改善。

> 不一样，要在码流送解码之前调用，把数据写入 ddr 让硬件真正访问到

@HermanChen  请教一下， camera_source 里面capture的buffer也是 dma，jpeg 解码器为什么需要同步呢？

tubzby, 2025-09-05:
另外在debug时发现，如果在jpeg解码之前 将 mjpeg 写入到文件，就不会有这个问题。
比如写50张mjpeg，则前面50个yuv没问题，后面的都会有问题。