rockchip-linux/mpp #633: mpp_init初始化失败求教

binruable, 2024-07-17:
mpp[1001]: mpp_info: mpp version: 41c9c809 author: sayon.chen    2020-12-24 [mpp2str]: Add rc_gop_mode 2 str
mpp[1001]: mpp_rt: NOT found ion allocator
mpp[1001]: mpp_rt: found drm allocator
mpp[1001]: mpp_platform: can not found match soc name: rockchip,rk3588s-evb4-lp4x-v10 rockchip,rk3588
mpp[1001]: mpp_serivce: set client type 0 failed
mpp[1001]: hal_h264d_api: mpp_dev_init failed ret: -1
mpp[1001]: mpp_hal: mpp_hal_init hal h264d_rkdec init failed ret -1
mpp[1001]: mpp_hal: mpp_hal_init could not found coding type 7
mpp[1001]: mpp_dec: mpp_dec_init could not init hal
mpp[1001]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[1001]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[1001]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[1001]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[1001]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[1001]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[1001]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[1001]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[1001]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[1001]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[1001]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[1001]: mpp: error found on mpp initialization
mpp[1001]: mpp_log: mpp_init failed

HermanChen, 2024-07-17:
author: sayon.chen 2020-12-24
版本太旧了，用 develop 分支

GDbbq, 2024-08-19:
我也遇到了这样的问题
[image]

HermanChen, 2024-08-21:
hal_h264d_api: mpp_dev_init failed ret: -1
这个说明内核驱动不匹配，跑下 mpp_platform_test 看看结果