rockchip-linux/mpp #634: mpp_rkvenc fdf40000.rkvenc: no memory region mapped

denghengli, 2024-07-18:
### 一、版本信息
平台：RK3568
kernel：5.10.198
libmpp：1.0.6
librga: rga_api version 1.3.1_[11] (RGA is compiling with meson base: $PRODUCT_BASE)
RGA Deriver：RGA2 Device Driver: v2.1.0

### 二、mpp使用说明
1、4路摄像头rtsp取流-->mpp H265解码-->RGA分辨率转换--->mppH65编码+OSD字符叠加
2、每路摄像头rtsp取流 跟 mpp编解码都是独立的线程，两个线程直接通过的队列传递AVPacket

### 三、日志信息
1、kernel启动的时候mpp_rkvenc相关的打印信息，看着有一些失败的信息
[    2.781678] mpp_rkvenc fdf40000.rkvenc: Adding to iommu group 5
[    2.781888] mpp_rkvenc fdf40000.rkvenc: probing start
[    2.782322] mpp_rkvenc fdf40000.rkvenc: bin=0
[    2.782344] mpp_rkvenc fdf40000.rkvenc: Failed to get leakage
[    2.782427] mpp_rkvenc fdf40000.rkvenc: pvtm = 87650, from nvmem
[    2.782437] mpp_rkvenc fdf40000.rkvenc: pvtm-volt-sel=1
[    2.782605] mpp_rkvenc fdf40000.rkvenc: avs=0
[    2.782718] mpp_rkvenc fdf40000.rkvenc: failed to find power_model node
[    2.782730] mpp_rkvenc fdf40000.rkvenc: failed to initialize power model
[    2.782738] mpp_rkvenc fdf40000.rkvenc: failed to get dynamic-coefficient
[    2.783315] mpp_rkvenc fdf40000.rkvenc: probing finish
[    2.783778] mpp_rkvdec2 fdf80200.rkvdec: Adding to iommu group 6
[    2.783995] mpp_rkvdec2 fdf80200.rkvdec: rkvdec, probing start
[    2.784312] mpp_rkvdec2 fdf80200.rkvdec: shared_niu_a is not found!
[    2.784330] rkvdec2_init:1022: No niu aclk reset resource define
[    2.784340] mpp_rkvdec2 fdf80200.rkvdec: shared_niu_h is not found!
[    2.784347] rkvdec2_init:1025: No niu hclk reset resource define
[    2.784530] mpp_rkvdec2 fdf80200.rkvdec: bin=0
[    2.784617] mpp_rkvdec2 fdf80200.rkvdec: leakage=58
[    2.784626] mpp_rkvdec2 fdf80200.rkvdec: leakage-volt-sel=0
[    2.784658] mpp_rkvdec2 fdf80200.rkvdec: pvtm = 87650, from nvmem
[    2.784667] mpp_rkvdec2 fdf80200.rkvdec: pvtm-volt-sel=1
[    2.784829] mpp_rkvdec2 fdf80200.rkvdec: avs=0
[    2.784945] mpp_rkvdec2 fdf80200.rkvdec: failed to find power_model node
[    2.784958] mpp_rkvdec2 fdf80200.rkvdec: failed to initialize power model
[    2.784966] mpp_rkvdec2 fdf80200.rkvdec: failed to get dynamic-coefficient
[    2.785085] mpp_rkvdec2 fdf80200.rkvdec: sram_start 0x00000000fdcc0000
[    2.785090] mpp_rkvdec2 fdf80200.rkvdec: rcb_iova 0x0000000010000000
[    2.785094] mpp_rkvdec2 fdf80200.rkvdec: sram_size 45056
[    2.785098] mpp_rkvdec2 fdf80200.rkvdec: rcb_size 65536
[    2.785104] mpp_rkvdec2 fdf80200.rkvdec: min_width 512
[    2.785147] mpp_rkvdec2 fdf80200.rkvdec: link mode probe finish
[    2.785215] mpp_rkvdec2 fdf80200.rkvdec: probing finish

### 2、运行一段时间后的dmesg输出
**挂测1：**
[ 1045.126621] rga2_mmu: RGA2 failed to get pmd, result = -14, pageCount = 771
[ 1045.126726] rga2_mmu: rga2 map dst memory failed
[ 1045.126738] rga2_reg_init, [797] set mmu info error
[ 1045.126744] rga2: init reg fail
[ 1045.174149] rk_vcodec: mpp_translate_reg_address:1816: reg[ 70]: 0xffffffff fd -1 failed
[ 1045.174328] rk_vcodec: mpp_task_dump_mem_region:2003: --- dump task 0 mem region ---
[ 1045.174376] mpp_rkvenc fdf40000.rkvenc: no memory region mapped
[ 1045.174431] rk_vcodec: mpp_process_task_default:614: alloc_task failed.
[ 1045.174501] rk_vcodec: mpp_wait_result_default:902: session 898:3 pending list is empty!
[ 1045.174519] rk_vcodec: mpp_msgs_wait:1612: session 3 wait result ret -5
[ 3060.075398] capability: warning: `vsftpd' uses 32-bit capabilities (legacy support in use)
**挂测2：**
[13036.093741] rk_vcodec: mpp_wait_result_default:910: wait task break by signal
[13036.093779] rk_vcodec: mpp_wait_result_default:910: wait task break by signal
[13036.093898] rk_vcodec: mpp_wait_result_default:910: wait task break by signal
[13036.093928] rk_vcodec: mpp_wait_result_default:910: wait task break by signal
[13036.093949] rk_vcodec: mpp_wait_result_default:910: wait task break by signal
[13036.094066] rk_vcodec: mpp_wait_result_default:910: wait task break by signal
[15490.793240] rkvdec2_link_wait_result:1191: wait task break by signal
[23511.264064] mpp_dma_import_fd:210: dma_buf_get fd 116 failed(-9)
[23511.264245] rk_vcodec: mpp_task_attach_fd:1760: can't import dma-buf 116
[23511.264267] rk_vcodec: mpp_translate_reg_address:1816: reg[133]: 0x00000074 fd 116 failed
[23511.264291] rk_vcodec: mpp_task_dump_mem_region:2003: --- dump task 0 mem region ---
[23511.264324] rk_vcodec: mpp_task_dump_mem_region:2008: reg[128]: 0x00000000fddb0000, size 80000
[23511.264343] rk_vcodec: mpp_task_dump_mem_region:2008: reg[129]: 0x00000000fddb0000, size 80000
[23511.264357] rk_vcodec: mpp_task_dump_mem_region:2008: reg[130]: 0x00000000fdc10000, size 195000
[23511.264374] rk_vcodec: mpp_task_dump_mem_region:2008: reg[131]: 0x00000000ff410000, size 10000
[23511.264389] rk_vcodec: mpp_task_dump_mem_region:2008: reg[132]: 0x00000000fa4b0000, size 195000
[23511.264428] rkvdec2_link_process_task:1134: alloc_task failed.
[23511.264952] rkvdec2_link_wait_result:1185: session 00000000e7033090 pending list is empty!
[23511.264993] rk_vcodec: mpp_msgs_wait:1612: session 631 wait result ret -5
[23511.269457] rga2_mmu: RGA2 failed to get pte, result = -14, pageCount = 338
[23511.269543] rga2_mmu: rga2 map src0 memory failed
[23511.269560] rga2: init reg fail
[23511.272238] rga2_mmu: RGA2 failed to get pmd, result = -14, pageCount = 771
[23511.272430] rga2_mmu: rga2 map dst memory failed
[23511.272475] rga2: init reg fail

### 四、问题疑问
1、kernel启动的时候的输出信息来看，是rkvdec驱动初始化有异常吗
2、运行一段时间后出现上面dmesg的问题，会是什么原因呢，应该怎么解决或避免？
3、如果长时间这样运行会有什么问题吗，是否有影响
4、RGA也会报错，不知道是否了解会是什么原因呢

HermanChen, 2024-07-23:
1. 内核启动 log 没什么问题，只是一些报警用 log
2. dmesg 看起来是 rga 先报错了 rga2 map dst memory failed，可能是 RGA 使用空间超 4G 空间了？或者是开了解码 FBC 输出，错误帧导致了 RGA 异常
3. 会有问题

moo-the-cow, 2024-08-30:
I have that issue too: https://github.com/Joshua-Riek/ubuntu-rockchip/issues/1002
I posted quite some infos about it, it is still unresolved

HermanChen, 2024-08-30:
It seems a kernel issue. Could you please provide the kernel mpp directory?

HermanChen, 2024-08-30:
@FumasterLin reg 133 is a rcb address?

moo-the-cow, 2024-08-30:
ok, I just upgraded the custom kernel of that repo from 1021 to 1025 it seems indeed a kernel issue, because it works now.
Thank you for the hint!

FumasterLin, 2024-08-30:
> @FumasterLin reg 133 is a rcb address?

是的，有点奇怪

denghengli, 2024-08-30:
> ok, I just upgraded the custom kernel of that repo from 1021 to 1025 it seems indeed a kernel issue, because it works now. Thank you for the hint!

Are you saying that upgrading the kernel from 6.1.0-1021 to 6.1.0-1025 will solve the problem? Does that mean that this issue will also occur on 5.10.198?