rockchip-linux/mpp #637: mpp buffer creating EGL image fails

MonocleSecurity, 2024-07-24:
Hardware is a RK3588 Ubuntu 22 board ROC-RK3588S-PC https://en.t-firefly.com/doc/download/142.html

My example code here fails on line 181 where I call eglCreateImage() and I believe this should work.

https://github.com/MonocleSecurity/rockchipexample/blob/main/main.cpp#L181

HermanChen, 2024-07-25:
EGL_DMA_BUF_PLANE1_OFFSET_EXT, hor_stride * height -> EGL_DMA_BUF_PLANE1_OFFSET_EXT, hor_stride * ver_stride
What is the frame resolution?

MonocleSecurity, 2024-07-25:
1280x720 is the original resolution. The stride is 1280x1280.

HermanChen, 2024-07-26:
What is the exactly decoder hor/ver_stride ouput?

MonocleSecurity, 2024-07-28:
`Creating EGL image: 1280 720 0 0 0 1280 720`

Horizontal Stride: 1280
Vertical Stride: 720

Having thought about it for a few days, I think the buffer may need to be converted to RGB888 for the EGL create image function to consume it.

MonocleSecurity, 2024-07-28:
Resolved, I needed to use:

`PFNEGLCREATEIMAGEKHRPROC eglCreateImageKHR = (PFNEGLCREATEIMAGEKHRPROC)eglGetProcAddress("eglCreateImageKHR");`

closed 2024-07-28 by MonocleSecurity