rockchip-linux/mpp #638: rk3588开发相关问题

lawaarch, 2024-07-26:
陈工您好：
        我有几个问题请教，说一下前提。我目前手里3588是讯为的，拉取最新的mpp,可以执行测试程序。
第一：我想在rk3588上面主要做编解码开发，我通过./mpi_dec_test 测试性能基本满足，是不是意味着我板子上面的rga库是可以正常使用的，之前在3568上面开发因为rga驱动问题踩过坑。

第二：我测试了./mpi_dec_multi_test 32路 12路但是我没有看到编码器的多线程测试，编码器是不支持多线程吗？

第三：我在测试mpp的时候拉取readme指定的rtsp代码，发现项目过旧，无法正常运行，望更新

第四：关于MPP编解码接口的二次开发，我能参考的MPP开发文档还是2020版本的(之前编解码接口初始化就失败了)，您这边有最新版的吗？mpp编解码二次开发主要参考代码除了MPP项目还有哪些可以参考的

HermanChen, 2024-07-27:
一、mpp 中不包括 rga 库，需要测试下，正常应该是可以的
二、多线程用 mpi_dec_mt_test
三、mpp 之外的库我们无法更新……用这个试试 https://github.com/nyanmisaka/ffmpeg-rockchip
四、库里的 doc 目录里有文档更新

lawaarch, 2024-07-31:
> 一、mpp 中不包括 rga 库，需要测试下，正常应该是可以的 二、多线程用 mpi_dec_mt_test 三、mpp 之外的库我们无法更新……用这个试试 https://github.com/nyanmisaka/ffmpeg-rockchip 四、库里的 doc 目录里有文档更新

好的谢谢 陈工

weiyebing, 2025-04-07:
您好请教下，我在使用mpp进行ffmpeg上推流是出现  mpp_platform: client 12 driver is not ready!这样的错误，这种情况应该怎么处理啊