rockchip-linux/mpp #639: 3588 编码1080 40Mbps 的mjpeg 系统崩溃

yukuilong, 2024-07-28:
Android13 
编码mjpeg 640*360 ，mjpeg 1280*720 都是正常的 ，但是编码1080分辨率时 系统就崩溃了，报错信息如下：
`
[  605.233827][    C0] rk_iommu fdbac800.iommu: Page fault at 0x00000000ffff8500 of type read
[  605.233875][    C0] rk_iommu fdbac800.iommu: iova = 0x00000000ffff8500: dte_index: 0x3ff pte_index: 0x3f8 page_offset: 0x500
[  605.233889][    C0] rk_iommu fdbac800.iommu: mmu_dte_addr: 0x0000000000201000 dte@0x0000000000201ffc: 0x4579f001 valid: 1 pte@0x000000004579ffe0: 0x000000 valid: 0 page@0x0000000000000000 flags: 0x0
[  605.233902][    C0] rk_iommu fdbac800.iommu: fault addr 0xffff8500 status 4b arg 000000002df2d1fe
[  605.233911][    C0] rk_vcodec: mpp_task_dump_mem_region:2017: --- dump task 3 mem region ---
[  605.233920][    C0] rk_vcodec: mpp_task_dump_mem_region:2023: reg[ 48]: 0x00000000ffd00000, size 2f8000
[  605.233929][    C0] rk_vcodec: mpp_task_dump_mem_region:2023: reg[ 49]: 0x00000000ffd00000, size 2f8000
[  605.233938][    C0] rk_vcodec: mpp_task_dump_mem_region:2023: reg[ 50]: 0x00000000ffd00000, size 2f8000
[  605.233946][    C0] rk_vcodec: mpp_task_dump_mem_region:2023: reg[ 77]: 0x00000000ff880000, size 7f000
[  605.233953][    C0] rk_vcodec: mpp_task_dump_hw_reg:2063: --- dump hardware register ---
[  605.233962][    C0] rk_vcodec: mpp_task_dump_hw_reg:2068: reg[000]: 0000: 0x00000000
[  605.233970][    C0] rk_vcodec: mpp_task_dump_hw_reg:2068: reg[001]: 0004: 0x00000000
[  605.233977][    C0] rk_vcodec: mpp_task_dump_hw_reg:2068: reg[002]: 0008: 0x070a141d
[  605.233985][    C0] rk_vcodec: mpp_task_dump_hw_reg:2068: reg[003]: 000c: 0x090e1a25
[  605.233992][    C0] rk_vcodec: mpp_task_dump_hw_reg:2068: reg[004]: 0010: 0x04060609
[  605.233999][    C0] rk_vcodec: mpp_task_dump_hw_reg:2068: reg[005]: 0014: 0x06080a0c
[  605.234006][    C0] rk_vcodec: mpp_task_dump_hw_reg:2068: reg[006]: 0018: 0x0f161f26
[  605.234013][    C0] rk_vcodec: mpp_task_dump_hw_reg:2068: reg[007]: 001c: 0x161a2327
[  605.234020][    C0] rk_vcodec: mpp_task_dump_hw_reg:2068: reg[008]: 0020: 0x0a0a1014
[  605.234028][    C0] rk_vcodec: mpp_task_dump_hw_reg:2068: reg[009]: 0024: 0x10171723
[  605.234036][    C0] rk_vcodec: mpp_task_dump_hw_reg:2068: reg[010]: 0028: 0x1b20292d
[  605.234043][    C0] rk_vcodec: mpp_task_dump_hw_reg:2068: reg[011]: 002c: 0x2c2a3028
[  605.234050][    C0] rk_vcodec: mpp_task_dump_hw_reg:2068: reg[012]: 0030: 0x14181c20
[  605.234058][    C0] rk_vcodec: mpp_task_dump_hw_reg:2068: reg[013]: 0034: 0x18161619
[  605.234065][    C0] rk_vcodec: mpp_task_dump_hw_reg:2068: reg[014]: 0038: 0x292d3029
[  605.234072][    C0] rk_vcodec: mpp_task_dump_hw_reg:2068: reg[015]: 003c: 0x1f252828
[  605.234081][    C0] rk_vcodec: mpp_task_dump_hw_reg:2068: reg[016]: 0040: 0x07070a13
[  605.234088][    C0] rk_vcodec: mpp_task_dump_hw_reg:2068: reg[017]: 0044: 0x07080a1a
[  605.234095][    C0] rk_vcodec: mpp_task_dump_hw_reg:2068: reg[018]: 0048: 0x28282828
[  605.234102][    C0] rk_vcodec: mpp_task_dump_hw_reg:2068: reg[019]: 004c: 0x28282828
[  605.234109][    C0] rk_vcodec: mpp_task_dump_hw_reg:2068: reg[020]: 0050: 0x0a0a1628
[  605.234118][    C0] rk_vcodec: mpp_task_dump_hw_reg:2068: reg[021]: 0054: 0x131a2828
[  605.234126][    C0] rk_vcodec: mpp_task_dump_hw_reg:2068: reg[022]: 0058: 0x28282828
[  605.234133][    C0] rk_vcodec: mpp_task_dump_hw_reg:2068: reg[023]: 005c: 0x28282828
[  605.234141][    C0] rk_vcodec: mpp_task_dump_hw_reg:2068: reg[024]: 0060: 0x28282828
[  605.234148][    C0] rk_vcodec: mpp_task_dump_hw_reg:2068: reg[025]: 0064: 0x28282828
[  605.234155][    C0] rk_vcodec: mpp_task_dump_hw_reg:2068: reg[026]: 0068: 0x28282828
[  605.234162][    C0] rk_vcodec: mpp_task_dump_hw_reg:2068: reg[027]: 006c: 0x28282828
[  605.234171][    C0] rk_vcodec: mpp_task_dump_hw_reg:2068: reg[028]: 0070: 0x28282828
[  605.234179][    C0] rk_vcodec: mpp_task_dump_hw_reg:2068: reg[029]: 0074: 0x28282828
[  605.234186][    C0] rk_vcodec: mpp_task_dump_hw_reg:2068: reg[030]: 0078: 0x28282828
[  605.234193][    C0] rk_vcodec: mpp_task_dump_hw_reg:2068: reg[031]: 007c: 0x28282828
[  605.234200][    C0] rk_vcodec: mpp_task_dump_hw_reg:2068: reg[032]: 0080: 0x00000000
[  605.234207][    C0] rk_vcodec: mpp_task_dump_hw_reg:2068: reg[033]: 0084: 0x00000000
[  605.234215][    C0] rk_vcodec: mpp_task_dump_hw_reg:2068: reg[034]: 0088: 0x00000000
[  605.234222][    C0] rk_vcodec: mpp_task_dump_hw_reg:2068: reg[035]: 008c: 0x00000000
[  605.234229][    C0] rk_vcodec: mpp_task_dump_hw_reg:2068: reg[036]: 0090: 0x00000000
[  605.234237][    C0] rk_vcodec: mpp_task_dump_hw_reg:2068: reg[037]: 0094: 0x00000000
[  605.234245][    C0] rk_vcodec: mpp_task_dump_hw_reg:2068: reg[038]: 0098: 0x00000000
[  605.234252][    C0] rk_vcodec: mpp_task_dump_hw_reg:2068: reg[039]: 009c: 0x00000000
`

HermanChen, 2024-07-29:
输入图像 stride 需要 16 对齐

yukuilong, 2024-07-29:
输入的分辨率是 1920*1080  的 ，wstride 是16 对齐的 ，hstride 也要16对齐吗

HermanChen, 2024-07-29:
是的，这版的老 jpeg 编码器需要水平垂直都 16 对齐，比较麻烦

yukuilong, 2024-07-29:
嗯 改成 16对齐可以了 ，感谢

closed 2024-07-29 by HermanChen