rockchip-linux/mpp #640: Color calibration rk3588.

wmport, 2024-07-28:
Color calibration is related to the mpp driver or mali610, nanopc-t6 rk3588.
Colors are not reproduced correctly in different modes ... examples below vp9 and av1

[image]

[image]

HermanChen, 2024-07-29:
it seems that HDR info is not processed correctly. How to reproduce it?

mo123, 2024-07-29:
Rockchip told me RK3588 doesn't support 10-bit AV1, since AV1 HDR needs 10-bit color, it will only display in 8-bit color.

wmport, 2024-07-29:
> похоже, что информация HDR обрабатывается неправильно. Как это воспроизвести?

https://github.com/Joshua-Riek/ubuntu-rockchip/discussions/638#discussioncomment-9244689