rockchip-linux/mpp #644: 用ffmpeg拉流，交给mpp解码h265一直frame is none（但解码264正常）

HouLingLXH, 2024-08-02:
我使用ffmpeg 拉取rtsp流，并将packet数据交给mpp进行解码，
当视频流为h264时，我设置解码为 MPP_VIDEO_CodingAVC ，解码成功；
当视频流为h265时，我设置解码为 MPP_VIDEO_CodingHEVC，解码失败：
   decode_get_frame获取的frame 一直是 none 

按理说从h264切换为h265解码应该只要修改MPP_VIDEO_CodingHEVC，其他的都不用动吧？ 为什么不对呢？
mpp版本是
   d378a28e author: daijh         2023-12-22 update version  (HEAD, m/firefly/ubuntu, firefly-linux/rk3588/firefly)

HermanChen, 2024-08-02:
是在什么芯片上

HouLingLXH, 2024-08-02:
3588

HermanChen, 2024-08-02:
那不应该，用 develop 最新试试看

HouLingLXH, 2024-08-02:
刚试了，还是不行

HouLingLXH, 2024-08-02:
我是用ffmpeg的av_read_frame 方法循环的获取到AVPacket 数据，然后把data和size传递给mpp解码器解码

大概流程是这样
   AVPacket pkt;
   av_read_frame(ctx_in,&pkt)  ....
   
   uint8_t* pkt_data =  pkt->data；
   int pkt_size  = pkt->size    ；
   mpp_packet_set_data(packet, pkt_data);
   mpp_packet_set_size(packet, pkt_size);
   mpp_packet_set_pos(packet, pkt_data);
   mpp_packet_set_length(packet, pkt_size);

   ret = mpi->decode_put_packet(ctx, packet);

在h264 时是正常的， h265就不行了。  请问还有哪里需要改吗？除了MPP_VIDEO_CodingHEVC

HermanChen, 2024-08-02:
不太了解，正常应该不需要了，存码流下来看看是不是 vps/sps/pps 什么的没送

HouLingLXH, 2024-08-02:
额，这个要怎么看啊？  我用vlc拉流， 还有用ffmpeg直接软解都是正常的

HouLingLXH, 2024-08-02:
这样会有问题吗：
hevc @ 0x560492798280] PPS id out of range: 0
    Last message repeated 1 times
[hevc @ 0x560492798280] Error parsing NAL unit #0.
[hevc @ 0x560492798280] PPS id out of range: 0
    Last message repeated 3 times
[hevc @ 0x560492798280] Error parsing NAL unit #0.
[hevc @ 0x560492798280] PPS id out of range: 0
[hevc @ 0x560492798280] Error parsing NAL unit #0.
[hevc @ 0x560492798280] PPS id out of range: 0
[hevc @ 0x560492798280] Error parsing NAL unit #0.
[hevc @ 0x560492798280] PPS id out of range: 0
    Last message repeated 1 times
[hevc @ 0x560492798280] Error parsing NAL unit #0.
[hevc @ 0x560492798280] PPS id out of range: 0
    Last message repeated 4 times
[hevc @ 0x560492798280] Error parsing NAL unit #0.
[hevc @ 0x560492798280] PPS id out of range: 0
[hevc @ 0x560492798280] Error parsing NAL unit #0.
[hevc @ 0x560492798280] PPS id out of range: 0

HermanChen, 2024-08-02:
说明给的码流里没有 vps/sps/pps，把头信息给丢了

HouLingLXH, 2024-08-03:
我是用 mediamtx + ffmpeg ，推送一个h265编码的MP4为 h265的 rtsp流，然后再拉流解码
使用vlc可以正常播放， 使用ffmpeg软解码方式可以获取图像帧。 
如果是h264的，mpp解码正常
但是h265的话，用mpp解码一直为none

换成从一个真实的h265录像机拉流， mpp解码可以正常

jjxyai, 2024-08-22:
我也遇到了，解析一个h265的MP4视频
[image]
log显示：H265D_PARSER: No start code is found.
不知道直接用ffmpeg [hw_decode](https://ffmpeg.org/doxygen/trunk/hw_decode_8c-example.html)来解码，不用这mpp的api了

wzw88486969, 2024-11-26:
DEFINE_RKMPP_DECODER(h264, H264, "h264_mp4toannexb,dump_extra")

DEFINE_RKMPP_DECODER(hevc, HEVC, "hevc_mp4toannexb,dump_extra")

DEFINE_RKMPP_DECODER(mpeg4, MPEG4, "dump_extra,mpeg4_unpack_bframes")

ffmpeg-rockchip里面的，要用后面的过滤 hevc_mp4toannexb....

Jialeen, 2024-12-09:
我用ffmpeg拉流给mpp解码，h264的，也无法成功解码，decode_get_frame这一步的时候返回的frame为0。各位大哥h264成功解码的可以给我参考参考吗。多谢多谢

HermanChen, 2024-12-10:
https://github.com/nyanmisaka/ffmpeg-rockchip
这个是非官方支持中最好的，用这个试试

nyanmisaka, 2024-12-10:
- MPP 解码需要 AnnexB 格式的码流，不支持 MP4/avcC 格式。所以需要先过一遍 `{h264,hevc}_mp4toannexb` 码流过滤器进行转换再才能给 `{h264,hevc}_rkmpp` 解码器解码。**FFmpeg 程序内部自动处理了这个过程，但通过 LibAV 接口直接调用的话需要自己手动处理码流过滤器这部分。**

- MPP 解码还需要 Sequence header。但 RTSP 视频的 Sequence header 是通过 SDP 传递的，H.26X 码流本身可能不带 Sequence header，所以需要 `dump_extra` 码流过滤器把 Sequence header 从 `AVCodecContext->extradata` 中解析出来作为一个 packet 先提交给 `{h264,hevc}_rkmpp` 解码器，才能开始解码。**同理，通过 LibAV 接口调用也需要自行处理这部分。**

closed 2025-03-03 by HouLingLXH