rockchip-linux/mpp #646: H264解码的输出内存拷贝

youyou0819, 2024-08-02:
解码器的输出内存用MPP_BUFFER_TYPE_DRM方式时，用rga的imcopy时部分情况下会提示拷贝失败，内存超过4G。
RgaBlit(1465) RGA_BLIT fail: Invalid argument
 RgaBlit(1466) RGA_BLIT fail: Invalid argument
fd-vir-phy-hnd-format[0, 0x7ea36a0000, (nil), 0, 0]
rect[0, 0, 2560, 1440, 2560, 1440, 2560, 0]
f-blend-size-rotation-col-log-mmu[0, 0, 0, 0, 0, 0, 1]
fd-vir-phy-hnd-format[0, 0x7f8c295000, (nil), 0, 0]
rect[0, 0, 2560, 1440, 2560, 1440, 2560, 0]
f-blend-size-rotation-col-log-mmu[0, 0, 0, 0, 0, 0, 1]
This output the user patamaters when rga call blit fail
srect[x,y,w,h] = [0, 0, 0, 0] src[w,h,ws,hs] = [2560, 1440, 2560, 1440]

drect[x,y,w,h] = [0, 0, 0, 0] dst[w,h,ws,hs] = [2560, 1440, 2560, 1440]

usage[0x80000]

dmesg打印信息：
[165234.642991] rga_mm: RGA_MMU unsupported memory larger than 4G!
[165234.643032] rga_mm: scheduler core[4] unsupported mm_flag[0x0]!
[165234.643108] rga_mm: rga_mm_map_buffer map virtual address error!
[165234.643118] rga_mm: job buffer map failed!
[165234.643126] rga_mm: src channel map job buffer failed!
[165234.643132] rga_mm: failed to map buffer
[165234.643147] rga_job: rga_job_commit: failed to map job info
[165234.643169] rga_job: request[2397982] task[0] job_commit failed.
[165234.643178] rga_job: rga request commit failed!
[165234.643188] rga: request[2397982] submit failed!

ncheng89, 2025-03-11:
这个问题有解决办法了吗，我也遇到了

youyou0819, 2025-03-12:
> 这个问题有解决办法了吗，我也遇到了

这是由于rga 2 和rga 3有个只支持4G以内的地址拷贝，可以申请内存的时候指定DMA32的内存或者拷贝指定RGA核支持4G外的地址运行即可。

Shine6Z, 2025-09-01:
试了指定内存空间，问题依旧，只是引发问题的时间变久了而已