rockchip-linux/mpp #647: Cannot get gstreamer to run on 6.1.0.x with RK3588 chipset (orange pi5+)

glowfishDE, 2024-08-06:
Hi,

when i try to run my gstreamer pipeline `gst-launch-1.0 v4l2src device=/dev/hdmirx ! identity name=ptsfixup signal-handoffs=TRUE ! identity drop-buffer-flags=GST_BUFFER_FLAG_DROPPABLE ! \
identity name=v_delay signal-handoffs=TRUE ! \
textoverlay text='' valignment=top halignment=right font-desc="Monospace, 5" name=overlay ! queue ! \
mpph265enc zero-copy-pkt=0 qp-max=51 gop=120 ! h265parse config-interval=-1 ! queue max-size-time=10000000000 max-size-buffers=1000 max-size-bytes=41943040 ! mux. \
alsasrc device=hw:CARD=rockchiphdmiin ! identity name=a_delay signal-handoffs=TRUE ! volume volume=1.0 ! audioconvert ! voaacenc bitrate=128000 ! aacparse ! queue max-size-time=10000000000 max-size-buffers=1000 ! mux. \
mpegtsmux name=mux ! appsink name=appsink`

i get the following errors:

`2024-08-06T10:18:27.076638+02:00 localhost belaUI[1055]: Gstreamer pipeline: gst-launch-1.0 v4l2src device=/dev/hdmirx ! identity name=ptsfixup signal-handoffs=TRUE ! identity drop-buffer-flags=GST_BUFFER_FLAG_DROPPABLE ! \
2024-08-06T10:18:27.076676+02:00 localhost belaUI[1055]: identity name=v_delay signal-handoffs=TRUE ! \
2024-08-06T10:18:27.076700+02:00 localhost belaUI[1055]:  queue ! \
2024-08-06T10:18:27.076722+02:00 localhost belaUI[1055]: mpph265enc zero-copy-pkt=0 qp-max=51 gop=120 ! h265parse config-interval=-1 ! queue max-size-time=10000000000 max-size-buffers=1000 max-size-bytes=41943040 ! mux. \
2024-08-06T10:18:27.076745+02:00 localhost belaUI[1055]: alsasrc device="hw:rockchiphdmiin" ! identity name=a_delay signal-handoffs=TRUE ! volume volume=1.0 ! audioconvert ! audioresample quality=10 sinc-filter-mode=1 ! opusenc bitrate=128000 ! opusparse ! queue max-size-time=10000000000 max-size-buffers=1000 ! mux. \
2024-08-06T10:18:27.076777+02:00 localhost belaUI[1055]: mpegtsmux name=mux ! appsink name=appsink
2024-08-06T10:18:27.516170+02:00 localhost mpp[1619]: mpp_info: mpp version: unknown mpp version for missing VCS info
2024-08-06T10:18:27.524202+02:00 localhost mpp[1619]: mpp_platform: client 12 driver is not ready!
2024-08-06T10:18:27.526729+02:00 localhost mpp[1619]: mpp_info: mpp version: unknown mpp version for missing VCS info
2024-08-06T10:18:27.529835+02:00 localhost mpp[1619]: message repeated 2 times: [ mpp_info: mpp version: unknown mpp version for missing VCS info]
2024-08-06T10:18:27.581125+02:00 localhost belaUI[1055]: Max bitrate: 8000000
2024-08-06T10:18:27.581179+02:00 localhost belaUI[1055]: Failed to get an encoder element from the pipeline, no dynamic bitrate control
2024-08-06T10:18:27.581203+02:00 localhost belaUI[1055]: A-V delay: 0 ms
2024-08-06T10:18:27.581226+02:00 localhost belaUI[1055]: Failed to get a ptsfixup element from the pipeline, not removing PTS jitter
`

glowfishDE, 2024-08-06:
Hi Herman,

is it possible to write in english? sorry :(

I translated what you wrote and i dont know what you mean by "check the parameters". Do you mean the gstreamer pipeline? I have included it. Or what parameters are you refering to?

I think the MPP1 is not working correctly on 6.1.0-1021-rockchip (Ubunut 24.04 LTS / Joshua Riek´s Image).

glowfishDE, 2024-08-06:
I also get alot of these errors:

g_object_set_is_valid_property: object class 'GstMppH265Enc' has no property named 'bitrate'

It worked with a 5.x Kernel just fine :(

HermanChen, 2024-08-07:
I have no idea about gst plugin implement. We need @JeffyCN to help on gst.

JeffyCN, 2024-08-07:
the plugin is using bps property instead of bitrate.

and that should be a warning instead of fatal error.

if it works well with kernel 5.10, maybe something wrong with the 6.10 kernel and mpp(driver/userspace), you can try to test with mpp's tests and gst videotestsrc.