rockchip-linux/mpp #651: mpp内部/分部分配的两个问题

circlefangzm, 2024-08-14:
大佬请教下，先谢谢了：
1. 如果是mpp半内部分配方式，把1080P NV12拷贝到系统内存（cpu需要处理），用哪个api？Rk3566上面大概需要多久？
2. 纯外部分配方式，drm类型，跟内部/半内部分配方式相比，解码会慢吗？会的话能慢多少？另外5.10的内核，说是要弄成cached的，具体怎么弄呢？没找到示例

HermanChen, 2024-08-15:
1. 看文档，在读取 buffer 的时候，先 mpp_buffer_sync 再读
2. 解码速度是一样的，不会慢，5.10 内核可以开 cache，但是 drm 的 cache 的内核接口可能有问题

circlefangzm, 2024-08-15:
谢谢您的及时回复。文档除了https://github.com/rockchip-linux/mpp/tree/develop/doc，还有别的吗？这里感觉还不不够详细。我的目的就是把解码后的帧通过网络发送出去（相当于只读），也得先sync出来是吧，这个sync_begin只是这对解码buffer吧？要在packet放入之前做吗？ 另外我这个目的直接用mpp_buffer_read不行吗？

HermanChen, 2024-08-15:
先 sync 是为了避免 cache 问题，如果是 uncache 的 buffer 就直接拷贝也可以

circlefangzm, 2024-08-15:
那默认情况下，半内部分配的buffer是cached还是uncached的呢？直接拷贝可以memcpy或者mpp_buffer_read都可以是吧?  还有如果需要sync那么什么时候begin什么时候end呢？

HermanChen, 2024-08-19:
看环境和 dmaheap 的配置