rockchip-linux/mpp #657: MPP rga blit报错

AlfredLeng, 2024-08-19:
```bash
[RKMEDIA][SYS][Debug]:Flow:VideoEncoderFlow set run_times to 1
[RKMEDIA][SYS][Debug]:Flow:FilterFlow:rkrga set run_times to 1
[RKMEDIA][VENC][Debug]:MPP Encoder: set osd data(3 regions) to frame
[RKMEDIA][SYS][Debug]:Flow:JpegLightEncoder set run_times to 1
[RKMEDIA][SYS][Debug]:Flow:VideoEncoderFlow set run_times to 1
[RKMEDIA][SYS][Debug]:Flow:FilterFlow:rkrga set run_times to 1
[RKMEDIA][SYS][Error]:FilterFlow:rkrga: buffer_pool get null buffer!
[RKMEDIA][VENC][Debug]:MPP Encoder: set osd data(3 regions) to frame
[RKMEDIA][SYS][Debug]:Flow:JpegLightEncoder set run_times to 1
[RKMEDIA][SYS][Debug]:Flow:VideoEncoderFlow set run_times to 1
[RKMEDIA][SYS][Debug]:Flow:FilterFlow:rkrga set run_times to 1
[RKMEDIA][VENC][Debug]:MPP Encoder: set osd data(3 regions) to frame
[RKMEDIA][SYS][Debug]:Argb8888_To_Region_Data Bitmap[384, 32] -> Canvas[384, 32], target=<384, 32>
[RKMEDIA][SYS][Debug]:Flow:JpegLightEncoder set run_times to 1
[RKMEDIA][SYS][Debug]:Flow:VideoEncoderFlow set run_times to 1
[RKMEDIA][SYS][Debug]:Flow:FilterFlow:rkrga set run_times to 1
[RKMEDIA][VENC][Debug]:MPP Encoder: config osd regions
[RKMEDIA][VENC][Debug]:MPP Encoder: setting osd regions...
[RKMEDIA][VENC][Debug]:MPP Encoder: Region[0] reuse old buff:12288, new_size:12288
[RKMEDIA][VENC][Debug]:MPP Encoder: set osd data(3 regions) to frame
[RKMEDIA][SYS][Debug]:Flow:JpegLightEncoder set run_times to 1
[RKMEDIA][SYS][Debug]:Flow:VideoEncoderFlow set run_times to 1
[RKMEDIA][SYS][Debug]:Flow:FilterFlow:rkrga set run_times to 1
[RKMEDIA][VENC][Debug]:MPP Encoder: set osd data(3 regions) to frame
[RKMEDIA][SYS][Debug]:Flow:JpegLightEncoder set run_times to 1
[RKMEDIA][SYS][Debug]:Flow:VideoEncoderFlow set run_times to 1
[RKMEDIA][SYS][Debug]:Flow:FilterFlow:rkrga set run_times to 1
[RKMEDIA][VENC][Debug]:MPP Encoder[JPEG]: config osd regions
[RKMEDIA][VENC][Debug]:MPP Encoder[OSD EX]: setting osd regions...
Error srcRect
[RgaBlit,723]Error srcRect

fd-vir-phy-hnd-format[160, (nil), (nil), (nil), 0]
rect[1024, 688, 320, 32, 1280, 720, 2560, 0]
f-blend-size-rotation-col-log-mmu[0, ff0501, 0, 0, 0, 0, 1]
fd-vir-phy-hnd-format[160, (nil), (nil), (nil), 0]
rect[1024, 688, 320, 32, 1280, 720, 2560, 0]
f-blend-size-rotation-col-log-mmu[0, 0, 0, 0, 0, 0, 1]
fd-vir-phy-hnd-format[185, (nil), (nil), (nil), 0]
rect[0, 0, 320, 32, 320, 32, 768, 0]
f-blend-size-rotation-col-log-mmu[0, 0, 0, 0, 0, 0, 1]
This output the user patamaters when rga call blit fail
[RKMEDIA][VENC][Error]:MPP Encoder[OSD EX]: do overlay failed!
[RKMEDIA][SYS][Debug]:Flow:JpegLightEncoder set run_times to 1
[RKMEDIA][SYS][Debug]:Flow:VideoEncoderFlow set run_times to 1
[RKMEDIA][SYS][Debug]:Flow:FilterFlow:rkrga set run_times to 1
[RKMEDIA][VENC][Debug]:MPP Encoder: set osd data(3 regions) to frame
err ws[1024,320,1280]

Error srcRect
[RgaBlit,723]Error srcRect

fd-vir-phy-hnd-format[160, (nil), (nil), (nil), 0]
rect[1024, 688, 320, 32, 1280, 720, 2560, 0]
f-blend-size-rotation-col-log-mmu[0, ff0501, 0, 0, 0, 0, 1]
fd-vir-phy-hnd-format[160, (nil), (nil), (nil), 0]
rect[1024, 688, 320, 32, 1280, 720, 2560, 0]
f-blend-size-rotation-col-log-mmu[0, 0, 0, 0, 0, 0, 1]
fd-vir-phy-hnd-format[185, (nil), (nil), (nil), 0]
rect[0, 0, 320, 32, 320, 32, 768, 0]
f-blend-size-rotation-col-log-mmu[0, 0, 0, 0, 0, 0, 1]
This output the user patamaters when rga call blit fail
[RKMEDIA][VENC][Error]:MPP Encoder[OSD EX]: do overlay failed!
[RKMEDIA][SYS][Debug]:Flow:JpegLightEncoder set run_times to 1
[RKMEDIA][SYS][Debug]:Flow:VideoEncoderFlow set run_times to 1
[RKMEDIA][SYS][Debug]:Flow:FilterFlow:rkrga set run_times to 1
[RKMEDIA][VENC][Debug]:MPP Encoder: set osd data(3 regions) to frame
err ws[1024,320,1280]
Error srcRect
[RgaBlit,723]Error srcRect

fd-vir-phy-hnd-format[160, (nil), (nil), (nil), 0]
rect[1024, 688, 320, 32, 1280, 720, 2560, 0]
f-blend-size-rotation-col-log-mmu[0, ff0501, 0, 0, 0, 0, 1]
```

```bash
[root@RV1126_RV1109:~]# cat /proc/mpp_service/version 
4cdfd254f099 author: Yandong Lin 2021-02-21 video: rockchip: mpp: Fix irq st
```

在1109上跑着出现这个，请问一下rga出错的输出怎么解读，什么情况下会导致的？

youyou0819, 2024-08-19:
查看下dmesg的信息

AlfredLeng, 2024-08-19:
额，出错当时的dmesg日志没有了断电重启了，首次发现看门狗复位时候的dmesg通过pstroe存下来
[2in-pstore.zip](https://github.com/user-attachments/files/16657054/2in-pstore.zip)
，不知道能不能作为参考