rockchip-linux/mpp #660: mpp硬件编码资源状态监控

Y-pandaman, 2024-08-20:
请问怎么实时监控rk3588的mpp编码的实时资源占用情况？
是使用`sudo cat /sys/kernel/debug/rknpu/load`·监控吗？

Y-pandaman, 2024-08-21:
> ![image](https://private-user-images.githubusercontent.com/86715812/359441184-db73a295-d17a-462d-ab45-314ec3fab598.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjQyMDM1NTYsIm5iZiI6MTcyNDIwMzI1NiwicGF0aCI6Ii84NjcxNTgxMi8zNTk0NDExODQtZGI3M2EyOTUtZDE3YS00NjJkLWFiNDUtMzE0ZWMzZmFiNTk4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MjElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODIxVDAxMjA1NlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTBjMjVjM2QxOTgxYzJjYTgxODgxNWFmMGQ1YmU5MWI5NmM1NzczZTY1MGUxMjlhNzczODBlM2I4YTY0ZDllMDcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.swW9-P52BWb4vwTLw1nhpWUyfW6fCLMFhvj5R98_hcU) 貌似只有这个

这个数字只是代表pid号

HermanChen, 2024-08-21:
目前编解码器还没有实现这个功能

closed 2024-08-21 by HermanChen