rockchip-linux/mpp #664: 重新源码编译rocketchip-mpp后缺失mppjpegenc和h264的mpp

huafengq, 2024-08-26:
在Ubuntu 22.04下，从的源码gstreamer-rockchip编译后，只有三个mpp编解码支持。麻烦帮忙看看为什么缺少了mppjpegenc和h264的mpp？谢谢！

root@dcic:/# gst-inspect-1.0 | grep mpp
rockchipmpp:  mppjpegdec: Rockchip's MPP JPEG image decoder
rockchipmpp:  mppvideodec: Rockchip's MPP video decoder
rockchipmpp:  mppvpxalphadecodebin: VP8/VP9 Alpha Decoder
typefindfunctions: audio/x-musepack: mpc, mpp, mp+
root@dcic:/# gst-inspect-1.0 mppjpegdec
Factory Details:
  Rank                     primary + 1 (257)
  Long-name                Rockchip's MPP JPEG image decoder
  Klass                    Decoder/Image
  Description              JPEG hardware decoder
  Author                   Randy Li <randy.li@rock-chips.com>, Jeffy Chen <jeffy.chen@rock-chips.com>

Plugin Details:
  Name                     rockchipmpp
  Description              Rockchip Mpp Video Plugin
  Filename                 /usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstrockchipmpp.so
  Version                  1.14.4
  License                  LGPL
  Source module            gst-rockchip
  Binary package           GStreamer Rockchip Plug-ins
  Origin URL               Unknown package origin

HermanChen, 2024-08-26:
@JeffyCN

JeffyCN, 2024-08-26:
mpp is not working, the plugins failed to init it.

neelsani, 2024-08-28:
> 在Ubuntu 22.04下，从的源码gstreamer-rockchip编译后，只有三个mpp编解码支持。麻烦帮忙看看为什么缺少了mppjpegenc和h264的mpp？谢谢！
> 
> root@dcic:/# gst-inspect-1.0 | grep mpp rockchipmpp: mppjpegdec: Rockchip's MPP JPEG image decoder rockchipmpp: mppvideodec: Rockchip's MPP video decoder rockchipmpp: mppvpxalphadecodebin: VP8/VP9 Alpha Decoder typefindfunctions: audio/x-musepack: mpc, mpp, mp+ root@dcic:/# gst-inspect-1.0 mppjpegdec Factory Details: Rank primary + 1 (257) Long-name Rockchip's MPP JPEG image decoder Klass Decoder/Image Description JPEG hardware decoder Author Randy Li [randy.li@rock-chips.com](mailto:randy.li@rock-chips.com), Jeffy Chen [jeffy.chen@rock-chips.com](mailto:jeffy.chen@rock-chips.com)
> 
> Plugin Details: Name rockchipmpp Description Rockchip Mpp Video Plugin Filename /usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstrockchipmpp.so Version 1.14.4 License LGPL Source module gst-rockchip Binary package GStreamer Rockchip Plug-ins Origin URL Unknown package origin

What kernel are you using?

huafengq, 2024-08-29:
> What kernel are you using?

The kernel is 5.10.198 for rk356x. the ubuntu 18.04 can work with mppjpegenc, mpph264enc/dec on the same kernel linux.
could you please help to point how to fix this issue, Thanks!

huafengq, 2024-08-29:
> mpp is not working, the plugins failed to init it.

Thanks! how to init the mpp success, or how to debug to find the root cause and where the initialization failed?

HermanChen, 2024-09-11:
provide the log please