rockchip-linux/mpp #670: 关于mpp_init失败的询问

FORGETSCHOLAR, 2024-09-05:
您好,
我的环境如下:
Linux firefly 5.10.160 #111 SMP Tue Jul 4 11:41:59 CST 2023 aarch64 aarch64 aarch64 GNU/Linux

我遇到的问题:
mpp_init的时候会发生段错误.

用gdb简单调试有这样的信息:
Program received signal SIGSEGV, Segmentation fault.
0x0000007ff786e974 in mpp_buffer_create () from /lib/aarch64-linux-gnu/librockchip_mpp.so.1
(gdb) bt
#0  0x0000007ff786e974 in mpp_buffer_create () from /lib/aarch64-linux-gnu/librockchip_mpp.so.1
#1  0x0000007ff7871c44 in mpp_buffer_get_with_tag () from /lib/aarch64-linux-gnu/librockchip_mpp.so.1
#2  0x0000007ff78e8940 in vdpu34x_h264d_init () from /lib/aarch64-linux-gnu/librockchip_mpp.so.1
#3  0x0000007ff78e8410 in hal_h264d_init () from /lib/aarch64-linux-gnu/librockchip_mpp.so.1
#4  0x0000007ff785bdac in mpp_hal_init () from /lib/aarch64-linux-gnu/librockchip_mpp.so.1
#5  0x0000007ff784e6a0 in mpp_dec_init () from /lib/aarch64-linux-gnu/librockchip_mpp.so.1
#6  0x0000007ff784713c in Mpp::init(MppCtxType, MppCodingType) () from /lib/aarch64-linux-gnu/librockchip_mpp.so.1
#7  0x0000007ff784b4f0 in mpp_init () from /lib/aarch64-linux-gnu/librockchip_mpp.so.1
#8  0x000000555555badc in MppDecoder::Init(int, int, void*) ()
#9  0x0000005555555b1c in main ()
(gdb) quit

查看dmesg | grep mpp,会有这样一段信息
[    5.713838] mpp_rkvenc2 fdbd0000.rkvenc-core: Adding to iommu group 10
[    5.713985] mpp_rkvenc2 fdbd0000.rkvenc-core: probing start
[    5.714096] mpp_rkvenc2 fdbd0000.rkvenc-core: dev_pm_opp_set_regulators: no regulator (venc) found: -19
[    5.714131] mpp_rkvenc2 fdbd0000.rkvenc-core: attach ccu as core 0
[    5.714231] mpp_rkvenc2 fdbd0000.rkvenc-core: probing finish
[    5.714288] mpp_rkvenc2 fdbe0000.rkvenc-core: Adding to iommu group 11
[    5.714442] mpp_rkvenc2 fdbe0000.rkvenc-core: probing start
[    5.714544] mpp_rkvenc2 fdbe0000.rkvenc-core: dev_pm_opp_set_regulators: no regulator (venc) found: -19
[    5.714599] mpp_rkvenc2 fdbe0000.rkvenc-core: attach ccu as core 1
[    5.714701] mpp_rkvenc2 fdbe0000.rkvenc-core: probing finish
[    5.715096] mpp_av1dec: Adding child /av1d@fdc70000
[    5.715153] mpp_av1dec: register device av1d-master
[    5.715175] mpp_av1dec av1d-master: av1_iommu_of_xlate,784
[    5.715215] mpp_av1dec av1d-master: Adding to iommu group 16
[    5.715396] mpp_av1dec av1d-master: probing start
[    5.715581] mpp_av1dec av1d-master: probing finish
[    5.715605] mpp_service mpp-srv: probe success

目前毫无头绪,恳请大佬解答

HermanChen, 2024-09-06:
跑下 mpp_platform_test，dmabuf 分配器用的 drm 还是 dmaheap？

FORGETSCHOLAR, 2024-09-06:
@HermanChen 
感谢回复，运行mpp_plat_test后，日志输出如下：
mpp_platform: client 12 driver is not ready!
mpp[424074]: mpp_platform: client 18 driver is not ready!
mpp[424074]: mpp_plat_test: kernel version: 5.10
mpp[424074]: mpp_plat_test: ioctl  version: mpp_service
mpp[424074]: mpp_plat_test: 
mpp[424074]: mpp_plat_test: chip name: rockchip,roc-rk3588s-pc rockchip,rk3588
mpp[424074]: mpp_plat_test: 
mpp[424074]: mpp_plat_test: chip vcodec type 1009221a
mpp[424074]: mpp_plat_test: found vpu2 codec
mpp[424074]: mpp_plat_test: found rkvdec decoder
mpp[424074]: mpp_plat_test: found rk jpeg decoder
mpp[424074]: mpp_plat_test: found rkvenc encoder
mpp[424074]: mpp_plat_test: 
mpp[424074]: mpp_plat_test: start probing decoder device name:
mpp[424074]: mpp_plat_test: H.264 decoder: /dev/mpp_service
mpp[424074]: mpp_plat_test: H.265 decoder: /dev/mpp_service
mpp[424074]: mpp_plat_test: MJPEG decoder: /dev/mpp_service
mpp[424074]: mpp_plat_test: VP9   decoder: /dev/mpp_service
mpp[424074]: mpp_plat_test: avs+  decoder: /dev/mpp_service
mpp[424074]: mpp_plat_test: 
mpp[424074]: mpp_plat_test: start probing encoder device name:
mpp[424074]: mpp_plat_test: H.264 encoder: /dev/mpp_service
mpp[424074]: mpp_plat_test: H.265 encoder: /dev/mpp_service
mpp[424074]: mpp_plat_test: MJPEG encoder: /dev/mpp_service
mpp[424074]: mpp_plat_test: mpp platform test done

dmabuf分配器，显示是使用drm

FORGETSCHOLAR, 2024-09-07:
> 跑下 mpp_platform_test，dmabuf 分配器用的 drm 还是 dmaheap？

mpp[7816]: mpp_dmabuf_test: mpp dmabuf test start
mpp[7816]: mpp_dma_heap: os_allocator_dma_heap_open open dma heap type 0 system failed!
mpp[7816]: mpp_buffer: MppBufferService get_group failed to get allocater with mode 0 type 20004
mpp[7816]: mpp_dmabuf_test: get dmaheap buffer group failed ret -1
mpp[7816]: mpp_dmabuf_test: mpp dmabuf test done failed

这是运行mpp_dmabuf_test之后的信息，请问我该如何解决这个问题？

HermanChen, 2024-09-11:
有 dma_heap 用 heap，然后再是有 drm 用 drm

HermanChen, 2024-09-11:
有没有 /dev/dma_heap 目录和下面的 heap 文件

FORGETSCHOLAR, 2024-09-11:
> 有没有 /dev/dma_heap 目录和下面的 heap 文件

是有这个目录的
firefly@firefly:~/niehaoyang$ ll -a /dev/dma_heap/
总用量 0
drwxr-xr-x  2 root root    160 1月   1  1970 ./
drwxr-xr-x 18 root root   4280 9月  15 01:27 ../
crw-------  1 root root 251, 4 9月   8 02:41 cma
crw-------  1 root root 251, 5 9月   8 02:41 cma-uncached
crw-------  1 root root 251, 0 9月   8 02:41 system
crw-------  1 root root 251, 1 9月   8 02:41 system-dma32
crw-------  1 root root 251, 2 9月   8 02:41 system-uncached
crw-------  1 root root 251, 3 9月   8 02:41 system-uncached-dma32

然后我运行mpi_dec_test日志是这样的：
运行mpi_dec_test有如下问题：
firefly@firefly:~/niehaoyang$ ./mpp/build/linux/aarch64/test/mpi_dec_test -t 7 -i /home/firefly/niehaoyang/rknn_yolov5_demo/build/build_linux_aarch64/rknn_yolov5_demo/street.h264
mpp[774424]: mpi_dec_utils: failed to open input file /home/firefly/niehaoyang/rknn_yolov5_demo/build/build_linux_aarch64/rknn_yolov5_demo/street.h264
mpp[774424]: mpi_dec_utils: cmd parse result:
mpp[774424]: mpi_dec_utils: input  file name: /home/firefly/niehaoyang/rknn_yolov5_demo/build/build_linux_aarch64/rknn_yolov5_demo/street.h264
mpp[774424]: mpi_dec_utils: output file name: 
mpp[774424]: mpi_dec_utils: width      :    0
mpp[774424]: mpi_dec_utils: height     :    0
mpp[774424]: mpi_dec_utils: type       :    7
mpp[774424]: mpi_dec_utils: max frames :    0
mpp[774424]: mpi_dec_test: mpi_dec_test start
mpp[774424]: mpp_info: mpp version: e9d95ce4 author: Yandong Lin   2024-09-06 fix[h265d]: fix output err causeby refs cleard
mpp[774424]: mpi_dec_test: 0x55a962db00 mpi_dec_test decoder test start w 0 h 0 type 7
mpp[774424]: mpp_dma_heap: os_allocator_dma_heap_open open dma heap type 0 system failed!
mpp[774424]: mpp_buffer: MppBufferService get_group failed to get allocater with mode 0 type 20001
mpp[774424]: mpp_buffer: mpp_buffer_group_limit_config input invalid group (nil)
mpp[774424]: mpp_platform: client 12 driver is not ready!
mpp[774424]: mpp_platform: client 18 driver is not ready!
mpp[774424]: mpp_dma_heap: os_allocator_dma_heap_open open dma heap type 0 system-uncached failed!
mpp[774424]: mpp_buffer: MppBufferService get_group failed to get allocater with mode 0 type 1
mpp[774424]: mpp_hal: mpp_hal_init hal h264d_rkdec init failed ret -1
mpp[774424]: mpp_hal: mpp_hal_init could not found coding type 7
mpp[774424]: mpp_dec: mpp_dec_init could not init hal
mpp[774424]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[774424]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[774424]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[774424]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[774424]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[774424]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[774424]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[774424]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[774424]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[774424]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[774424]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[774424]: mpp: error found on mpp initialization
mpp[774424]: mpi_dec_test: 0x55a962db00 mpp_init failed
mpp[774424]: mpi_dec_test: test failed ret -1
mpi_dec_test[774424]: mpp_mem_pool: put_pool found 1 used buffer size 304
mpi_dec_test[774424]: mpp_mem_pool: put_pool found 4 used buffer size 224

请问要如何解决

FORGETSCHOLAR, 2024-09-11:
> 有 dma_heap 用 heap，然后再是有 drm 用 drm

需要我在哪里显示指定用哪个分配器吗？

HermanChen, 2024-09-11:
权限，su 之后再运行

FORGETSCHOLAR, 2024-09-11:
> 权限，su 之后再运行

非常感谢，问题已经全部解决

closed 2024-09-11 by HermanChen