rockchip-linux/mpp #673: Hardware Accelerated Deinterlacing Support

jesus2k21, 2024-09-09:
I have been looking for the capability of deinterlacing video with GStreamer on rk3588 and have not found much on the topic. Are there such capabilities for hardware accelerated deinterlacing using GStreamer?

The MPP development guide seems to indicate that the platform is capable of deinterlacing but I have not found any way to do so using GStreamer.

nyanmisaka, 2024-09-10:
[`MPP_DEC_SET_ENABLE_DEINTERLACE,     /* MPP enable deinterlace by default. Vpuapi can disable it */`](https://github.com/search?q=repo%3Arockchip-linux%2Fmpp%20MPP_DEC_SET_ENABLE_DEINTERLACE&type=code)

It is automatically enabled as part of the decoder. Maybe there is an option in GStreamer that disables it?

jesus2k21, 2024-09-10:
I've been using the below pipeline with the environment variables active to try deinterlacing. I have a single thread, which is the deinterlacing thread, reaching 100% which tells me that MPP isn't being used to deinterlace.

I've been exploring the dmesg logs and found the following mpp-iep2 logs that may indicate some failure to allocate memory but the kernel code responsible for this is too complex for my understanding:
```
[    4.280521] mpp-iep2 fdbb0000.iep: Adding to iommu group 9
[    4.280610] mpp-iep2 fdbb0000.iep: probe device
[    4.280723] mpp-iep2 fdbb0000.iep: allocate roi buffer failed
[    4.280842] mpp-iep2 fdbb0000.iep: probing finish
```

I can confirm that my dtsi file has the following which I believe should be enabling the iep:
```
&iep {
	status = "okay";
};

&iep_mmu {
	status = "okay";
};
```

HermanChen, 2024-09-11:
> [`MPP_DEC_SET_ENABLE_DEINTERLACE, /* MPP enable deinterlace by default. Vpuapi can disable it [...]

Modify the gst plugin. It is not in mpp.

> I've been using the below pipeline with the environment variables active to try deinterlacing. I have a single thread, which is the deinterlacing thread, reaching 100% which tells me that MPP isn't being used to deinterlace.
> 
> ```
> export GST_MPP_NO_RGA=0
> export GST_VIDEO_CONVERT_USE_RGA=1
> gst-launch-1.0 -v -e \
> 	uridecodebin uri=file:///home/user/Documents/1080i.mkv \
> 	! deinterlace \
> 	! videoconvert ! video/x-raw,format=NV12,colorimetry=bt709 \
> 	! videorate ! video/x-raw,framerate=30/1 \
> 	! mpph264enc \
> 	! h264parse \
> 	! matroskamux \
> 	! filesink location=/home/user/Documents/out.mp4
> ```
> 
> I've been exploring the dmesg logs and found the following mpp-iep2 logs that may indicate some failure to allocate memory but the kernel code responsible for this is too complex for my understanding:
> 
> ```
> [    4.280521] mpp-iep2 fdbb0000.iep: Adding to iommu group 9
> [    4.280610] mpp-iep2 fdbb0000.iep: probe device
> [    4.280723] mpp-iep2 fdbb0000.iep: allocate roi buffer failed
> [    4.280842] mpp-iep2 fdbb0000.iep: probing finish
> ```
> 
> I can confirm that my dtsi file has the following which I believe should be enabling the iep:
> 
> ```
> &iep {
> 	status = "okay";
> };
> 
> &iep_mmu {
> 	status = "okay";
> };
> ```

busy query on loop? Add some sleep on the loop?

jesus2k21, 2024-09-11:
Just to clarify, which GStreamer plugin are you suggesting to modify? The plugin that I'm attempting to use to deinterlace?

HermanChen, 2024-09-11:
@JeffyCN

JeffyCN, 2024-09-11:
deinterlace is in mpp(using iep)

jesus2k21, 2024-09-12:
I appreciate everyones comments, I have decided to pursue other paths.

closed 2024-09-12 by jesus2k21