rockchip-linux/mpp #676: 264解码，纯外部内存分配，外部内存的fd，与decode_get_frame解析出来的fd不一致。

duwu891229, 2024-09-13:
#完全使用外部内存
1新建外部内存组mpp_buffer_group_get_external
2提交一些列外部内存到此组mpp_buffer_commit，其中有一些列fd，比如10-13
3解码获取decode_get_frame，mpp_frame_get_buffer，mpp_buffer_get_fd，得到一些列fd，比如14-17
4后续会用fd，去做RGA的yuv到rgb转换

其中1和3的fd，完全错开，感觉根本没使用提交的外部内存？

#完全使用内部内存
其中4会出现RGA无法访问4G内存的错误？

HermanChen, 2024-09-13:
commit 的 10-13 算是外部 buffer，这个引用是外部使用的
14-17 是解码器内部使用的，可以用 index 和外部 buffer 一一对应

4G 空间要在 buffer group 加 dma32 选项

duwu891229, 2024-09-13:
谢谢。

#外部内存的一帧的fd，与，编码器得到的一帧的fd，通过index对应。
一个进程空间内，2个不同的fd可以对应对同一块物理内存，（不同的vir_addr可以对应同一块物理内存）有点怪而已。

#buffer group 设定dma32不会，也不细究了。

HermanChen, 2024-09-13:
是的，会有点怪，主要是引用管理的问题，不然很容易出现外部把 buffer 一删，MPP 里就崩溃的情况，所以 MPP 内部会生成和保留一个独立引用

dma32 就是这个 https://github.com/rockchip-linux/mpp/blob/05b809393c41120c81fbdb39356b01c92aa77907/inc/mpp_buffer.h#L163