rockchip-linux/mpp #679: mpp解码h265报错，内核信息表示mpp重启

chaoxiToDo, 2024-09-14:
在我通过mpi_dec_test解码H265文件时出现如下报错
[image]
同时内核信息出现如下报错
[image]
（问题稳定复现，且每隔1200帧左右会继续报这个错误）请问这是什么原因，感谢！

HermanChen, 2024-09-14:
码流有错，用 ffmpeg 验证下码流是不是有问题

chaoxiToDo, 2024-09-14:
用ffplay可以正常播放，是一段40秒的视频。
这个是用ffmpeg检查的结果
[image]
这个是用StreamEye软件分析的结果，这里也有提示的信息，请问mpp的重启是跟这些问题有关吗，感谢！
[image]