rockchip-linux/mpp #685: 关于输入帧率与输出帧率不相等的情况下，mpp编码库一直打印 mpp_buffer: check buffer found NULL pointer from get_packet的问题

taozw-a, 2024-09-20:
大佬们，请教一下，使用mpp编码功能，编码器配置输入帧率为30，输出帧率为10，为了达到输入30帧，只需要输出10帧的抽稀效果，按照demo配置了参数rc:fps_in_num为30，rc:fps_out_num为10，结果是有效的，成功达到了抽稀的效果，但是mpp库日志却一直打印“mpp_buffer: check buffer found NULL pointer from get_packet”的信息，个数刚好是20个，请问这个打印要如何才能避免，这个打印是否是有问题？mpp库版本为：mpp_info: mpp version: 86cf93a0 author: Yanjun Liao   2024-07-27 fix[265e]:Fix the st refernce frame err in tsvc。

HermanChen, 2024-09-23:
这个打印是 get_packet 函数里打出来的，加点 log 分析下是哪里打印的

taozw-a, 2024-09-24:
> 这个打印是 get_packet 函数里打出来的，加点 log 分析下是哪里打印的

        MppPacketImpl *impl = (MppPacketImpl *)pkt;
        MppBuffer buf = impl->buffer;
        RK_U32 offset = (RK_U32)((char *)impl->pos - (char *)impl->data);

        mpp_buffer_sync_ro_partial_begin(buf, offset, impl->length);

        mpp_dbg_pts("%p output packet pts %lld\n", this, impl->pts);
是在 mpp_buffer_sync_ro_partial_begin(buf, offset, impl->length)这个函数里面打印的，里面判定了buf为空，但是这个buf是MppPacket *packet类型，是调用接口的是传进去的，编码器配置rc:fps_out_num为30的时候就不会报这个错误，一旦小于30就会出现。

HermanChen, 2024-09-24:
哦，这个是正常的，因为 30fps 输入，10 fps 输出的话，会有 20 帧的无数据的空 packet 输出来标记要释放输入图像
这种取出来的 MppBuffer 为空的 packet 忽略并 deinit 就好

taozw-a, 2024-09-24:
> 哦，这个是正常的，因为 30fps 输入，10 fps 输出的话，会有 20 帧的无数据的空 packet 输出来标记要释放输入图像 这种取出来的 MppBuffer 为空的 packet 忽略并 deinit 就好

那要怎么屏蔽这个打印呢，因为日志记录都是记录在系统日志的，打印太频繁会导致很难查看其它的调试日志，而且既然是正常的，做这个判断是不是多余了，有不可以直接屏蔽判断的语句

taozw-a, 2024-09-24:
> 哦，这个是正常的，因为 30fps 输入，10 fps 输出的话，会有 20 帧的无数据的空 packet 输出来标记要释放输入图像 这种取出来的 MppBuffer 为空的 packet 忽略并 deinit 就好

而且使用mpi_enc_test测试，是没有这种打印的，我这边排查了很多参数，也都跟demo的一致

HermanChen, 2024-09-24:
mpi_enc_test 用的是 block 模式，输入接口返回时输入图像即可释放，但如果是 non-block 模式，需要在输出端带的 meta 里才能知道要释放的输入图像，对比下 mpi_enc_test 和 mpi_ent_mt_test 在初始化和释放输入图像时的不同

HermanChen, 2024-09-24:
判断下 mpp_packet_get_buffer 的判断值啊

taozw-a, 2024-09-24:
好的，感谢大神指导

closed 2024-09-26 by taozw-a