rockchip-linux/mpp #689: 关于MppBufferGroup组释放内存释放不掉的问题

taozw-a, 2024-09-25:
我这边调用释放组内存会打印错误信息提示有数据未释放：
Aug  1 13:55:03 hi-target user.info mpp[9075]: mpp_info: mpp version: 86cf93a0 author: Yanjun Liao   2024-07-27 fix[265e]:Fix the st refernce frame err in tsvc
Aug  1 13:55:03 hi-target user.err mpp[9075]: mpp_platform: client 12 driver is not ready!
Aug  1 13:55:03 hi-target user.info mpp[9075]: mpp_enc: MPP_ENC_SET_RC_CFG bps 7776000 [0 : 0] fps [30:30] gop 10
Aug  1 13:55:04 hi-target user.err mpp[9075]: mpp_buffer: mpp_group 0x1a009e8 tag unknown caller CbbEncoderInit mode internal type drm deinit with 6266880 bytes not released
Aug  1 13:55:04 hi-target user.info mpp[9075]: mpp_buffer:  dumping buffer group 0x1a009e8 id 1 from mpp_buffer_group_deinit
Aug  1 13:55:04 hi-target user.info mpp[9075]: mpp_buffer: mode internal
Aug  1 13:55:04 hi-target user.info mpp[9075]: mpp_buffer: type drm
Aug  1 13:55:04 hi-target user.info mpp[9075]: mpp_buffer: limit size 0 count 0
Aug  1 13:55:04 hi-target user.info mpp[9075]: mpp_buffer: used buffer count 1
Aug  1 13:55:04 hi-target user.info mpp[9075]: mpp_buffer: buffer 0x1a00ea8 fd   13 size    3133440 ref_count  30 discard 0 caller CbbEncoderInit
Aug  1 13:55:04 hi-target user.info mpp[9075]: mpp_buffer: unused buffer count 0
Aug  1 13:55:04 hi-target user.err mpp[9075]: mpp_buffer: mpp_buffer_ref_dec found non-positive ref_count 0 caller CbbStartCapture
Aug  1 13:55:04 hi-target user.err mpp[9075]: mpp_buffer: mpp_buffer_ref_dec found non-positive ref_count 0 caller CbbStartCapture
Aug  1 13:55:04 hi-target user.err mpp[9075]: mpp_buffer: mpp_buffer_ref_dec found non-positive ref_count 0 caller CbbStartCapture
Aug  1 13:55:04 hi-target user.err mpp[9075]: mpp_buffer: mpp_buffer_ref_dec found non-positive ref_count 0 caller CbbStartCapture
Aug  1 13:55:04 hi-target user.info mpp[9075]: mpp_buffer: ~MppBufferService cleaning misc group
Aug  1 13:55:04 hi-target user.info mpp[9075]: mpp_buffer: ~MppBufferService cleaning leaked buffer
Aug  1 13:55:04 hi-target user.err cbb_camera_demo[9075]: mpp_mem_pool: put_pool found 30 used buffer size 304

用的rk3588，mpp版本是mpp version: 86cf93a0 author: Yanjun Liao
我这边MppBuffer申请流程是这样的：
       //申请组
        ret = mpp_buffer_group_get_internal(&p_encoder_handle->buf_group, MPP_BUFFER_TYPE_DRM | MPP_BUFFER_FLAGS_CACHABLE);
        //申请frame buffer
        ret = mpp_buffer_get(p_encoder_handle->buf_group, &p_encoder_handle->frame_buffer, p_encoder_handle->frame_size);
        //申请packet buffer
        ret = mpp_buffer_get(p_encoder_handle->buf_group, &p_encoder_handle->packet_buffer, p_encoder_handle->frame_size);
释放流程是这样的：
 if (p_encoder_handle->ctx) {
        p_encoder_handle->mpi->reset(p_encoder_handle->ctx);
        mpp_destroy(p_encoder_handle->ctx);
        p_encoder_handle->ctx = NULL;
    }

    if (p_encoder_handle->frame_buffer) {
        mpp_buffer_put(p_encoder_handle->frame_buffer);
        p_encoder_handle->frame_buffer = NULL;
    }

    if (p_encoder_handle->packet_buffer) {
        mpp_buffer_put(p_encoder_handle->packet_buffer);
        p_encoder_handle->packet_buffer = NULL;
    }

    if (p_encoder_handle->buf_group) {
        mpp_buffer_group_put(p_encoder_handle->buf_group);
        p_encoder_handle->buf_group = NULL;
    }
我这个流程是完成按照demo和手册的说明进行释放的，但是就是会打印这些信息，请大佬帮忙看看有没有问题

taozw-a, 2024-09-26:
> 我这边调用释放组内存会打印错误信息提示有数据未释放： Aug 1 13:55:03 hi-target user.info mpp[9075]: mpp_info: mpp version: [86cf93a](https://github.com/rockchip-linux/mpp/commit/86cf93a0280669d5f1ec32128f1b9af69a453e80) author: Yanjun Liao 2024-07-27 fix[265e]:Fix the st refernce frame err in tsvc Aug 1 13:55:03 hi-target user.err mpp[9075]: mpp_platform: client 12 driver is not ready! Aug 1 13:55:03 hi-target user.info mpp[9075]: mpp_enc: MPP_ENC_SET_RC_CFG bps 7776000 [0 : 0] fps [30:30] gop 10 Aug 1 13:55:04 hi-target user.err mpp[9075]: mpp_buffer: mpp_group 0x1a009e8 tag unknown caller CbbEncoderInit mode internal type drm deinit with 6266880 bytes not released Aug 1 13:55:04 hi-target user.info mpp[9075]: mpp_buffer: dumping buffer group 0x1a009e8 id 1 from mpp_buffer_group_deinit Aug 1 13:55:04 hi-target user.info mpp[9075]: mpp_buffer: mode internal Aug 1 13:55:04 hi-target user.info mpp[9075]: mpp_buffer: type drm Aug 1 13:55:04 hi-target user.info mpp[9075]: mpp_buffer: limit size 0 count 0 Aug 1 13:55:04 hi-target user.info mpp[9075]: mpp_buffer: used buffer count 1 Aug 1 13:55:04 hi-target user.info mpp[9075]: mpp_buffer: buffer 0x1a00ea8 fd 13 size 3133440 ref_count 30 discard 0 caller CbbEncoderInit Aug 1 13:55:04 hi-target user.info mpp[9075]: mpp_buffer: unused buffer count 0 Aug 1 13:55:04 hi-target user.err mpp[9075]: mpp_buffer: mpp_buffer_ref_dec found non-positive ref_count 0 caller CbbStartCapture Aug 1 13:55:04 hi-target user.err mpp[9075]: mpp_buffer: mpp_buffer_ref_dec found non-positive ref_count 0 caller CbbStartCapture Aug 1 13:55:04 hi-target user.err mpp[9075]: mpp_buffer: mpp_buffer_ref_dec found non-positive ref_count 0 caller CbbStartCapture Aug 1 13:55:04 hi-target user.err mpp[9075]: mpp_buffer: mpp_buffer_ref_dec found non-positive ref_count 0 caller CbbStartCapture Aug 1 13:55:04 hi-target user.info mpp[9075]: mpp_buffer: ~MppBufferService cleaning misc group Aug 1 13:55:04 hi-target user.info mpp[9075]: mpp_buffer: ~MppBufferService cleaning leaked buffer Aug 1 13:55:04 hi-target user.err cbb_camera_demo[9075]: mpp_mem_pool: put_pool found 30 used buffer size 304
> 
> 用的rk3588，mpp版本是mpp version: [86cf93a](https://github.com/rockchip-linux/mpp/commit/86cf93a0280669d5f1ec32128f1b9af69a453e80) author: Yanjun Liao 我这边MppBuffer申请流程是这样的： //申请组 ret = mpp_buffer_group_get_internal(&p_encoder_handle->buf_group, MPP_BUFFER_TYPE_DRM | MPP_BUFFER_FLAGS_CACHABLE); //申请frame buffer ret = mpp_buffer_get(p_encoder_handle->buf_group, &p_encoder_handle->frame_buffer, p_encoder_handle->frame_size); //申请packet buffer ret = mpp_buffer_get(p_encoder_handle->buf_group, &p_encoder_handle->packet_buffer, p_encoder_handle->frame_size); 释放流程是这样的： if (p_encoder_handle->ctx) { p_encoder_handle->mpi->reset(p_encoder_handle->ctx); mpp_destroy(p_encoder_handle->ctx); p_encoder_handle->ctx = NULL; }
> 
> ```
> if (p_encoder_handle->frame_buffer) {
>     mpp_buffer_put(p_encoder_handle->frame_buffer);
>     p_encoder_handle->frame_buffer = NULL;
> }
> 
> if (p_encoder_handle->packet_buffer) {
>     mpp_buffer_put(p_encoder_handle->packet_buffer);
>     p_encoder_handle->packet_buffer = NULL;
> }
> 
> if (p_encoder_handle->buf_group) {
>     mpp_buffer_group_put(p_encoder_handle->buf_group);
>     p_encoder_handle->buf_group = NULL;
> }
> ```
> 
> 我这个流程是完成按照demo和手册的说明进行释放的，但是就是会打印这些信息，请大佬帮忙看看有没有问题

已找到解决办法。

closed 2024-09-26 by taozw-a

zen010101, 2025-02-12:
你找到的是什么办法能否分享一下？