rockchip-linux/mpp #695: 请问一下，mpp的OSD 如何添加？

androidHQJ, 2024-10-14:


HermanChen, 2024-10-15:
参考 mpi_enc_test

androidHQJ, 2024-10-15:
> 参考 mpi_enc_test

osd动态添加中文字符 是不是需要 Freetype 生成 水印位图内容，然后再通过 mpp osd_data 加进去？

HermanChen, 2024-10-15:
是这样的

androidHQJ, 2024-10-15:
[image]

请教一下，这一块代码是什么意思？不清楚这些参数的作用

HermanChen, 2024-10-15:
osd 配置是一块一块的窗口，然后每个窗口是一个 region

androidHQJ, 2024-10-15:
> > 参考 mpi_enc_test
> 
> osd动态添加中文字符 是不是需要 Freetype 生成 水印位图内容，然后再通过 mpp osd_data 加进去？

现在窗口创建好了，就卡在怎么把 通过freetype+sdl 生成相应的bmp图叠加进去。我查了一些资料，看到rv1109/rv1126的OSD叠加可以通过 如图所示的方式。所以请教一下，rkmpp 是怎么去叠加的呢？
[image]

HermanChen, 2024-10-15:
这个函数里面也是调用 mpp 的接口

androidHQJ, 2024-10-17:
> 这个函数里面也是调用 mpp 的接口

现在定位到接口了：
mpp_meta_set_ptr(meta, KEY_OSD_DATA, (void*)&p->osd_data);
不太清楚  freetype 的到的位图 FT_Bitmap 怎么叠加到 osd_data中? 最好有个思路或者demo 更好..

denghengli, 2024-10-21:
这个可以通过查询字库的方式来生成字符的点阵自摸

HermanChen, 2024-10-21:
这个需要外部应用自己来处理了，mpp 库不支持这样的功能

androidHQJ, 2024-10-22:
> 这个可以通过查询字库的方式来生成字符的点阵自摸

是通过 freetype 来实现吗？

androidHQJ, 2024-10-25:
> 这个可以通过查询字库的方式来生成字符的点阵自摸

位图生成好了，不清楚如何叠加到 osd窗口上?

HermanChen, 2024-10-25:
用 rga 来叠加吧，这样比较通用一些，编码器的 osd 功能要看芯片，有的有，有的没有

androidHQJ, 2024-10-25:
好的，我来试试。感谢解答~

androidHQJ, 2024-10-25:
> 用 rga 来叠加吧，这样比较通用一些，编码器的 osd 功能要看芯片，有的有，有的没有

请教一下，rga 的api: imosd  rk3568不支持，所以，是用rga哪个api来叠加位图 字库生成的位图?

HermanChen, 2024-10-25:
https://github.com/airockchip/librga 看下这个，看对应的输入输出位图格式吧，比如把 osd 的 rgb 图叠到 yuv 或者 rga 图上

androidHQJ, 2024-10-25:
> https://github.com/airockchip/librga 看下这个，看对应的输入输出位图格式吧，比如把 osd 的 rgb 图叠到 yuv 或者 rga 图上

好的。学习了~

duiniuluantanqin, 2025-08-26:
> > https://github.com/airockchip/librga 看下这个，看对应的输入输出位图格式吧，比如把 osd 的 rgb 图叠到 yuv 或者 rga 图上
> 
> 好的。学习了~

请问搞懂了吗？先用freetype将文字生成rgb图，然后再用imblend 叠加到原图上？

duiniuluantanqin, 2025-08-27:
@HermanChen 先用freetype将文字生成rgb图，然后再用imblend 叠加到原图，无法实现自动反色功能？请问有什么解决方案吗？