rockchip-linux/mpp #697: mpi_->decode_put_packet(mpp_ctx_, mpp_pkt_);  Failed to put packet into MPP, error code: -1012

HuKai97, 2024-10-15:
想实现的功能：使用mmp代替cv::Mat image = cv::imdecode(cv::Mat(1, buf.bytesused, CV_8UC1, buffers_[buf.index].start), cv::IMREAD_COLOR);  减少cpu使用率

代码：
#include <ros/ros.h>
#include <sensor_msgs/Image.h>
#include <linux/videodev2.h>
#include <fcntl.h>
#include <sys/ioctl.h>
#include <sys/mman.h>
#include <unistd.h>
#include <cstring>
#include <vector>
#include <opencv2/opencv.hpp>
#include <rockchip/mpp_buffer.h>
// #include <rockchip/mpp_dec.h>
#include <rockchip/mpp_frame.h>
#include <rockchip/mpp_packet.h>
#include <rockchip/rk_mpi.h>

#define HEIGHT 480
#define WIDTH 640
#define FPS 30

class V4L2CameraNode
{
public:
    V4L2CameraNode() : mpp_ctx_(nullptr), mpp_pkt_(nullptr), mpp_frm_(nullptr), mpp_buf_grp_(nullptr)
    {
        fd_ = open("/dev/video0", O_RDWR);
        if (fd_ == -1)
        {
            ROS_ERROR("Failed to open V4L2 device");
            return;
        }

        if (initializeMPP() != MPP_OK)
        {
            ROS_ERROR("Failed to initialize MPP");
            close(fd_);
            return;
        }

        struct v4l2_capability cap;
        if (ioctl(fd_, VIDIOC_QUERYCAP, &cap) == -1)
        {
            ROS_ERROR("Failed to query V4L2 capabilities");
            close(fd_);
            return;
        }

        struct v4l2_format fmt;
        fmt.type = V4L2_BUF_TYPE_VIDEO_CAPTURE;
        fmt.fmt.pix.width = WIDTH;
        fmt.fmt.pix.height = HEIGHT;
        fmt.fmt.pix.pixelformat = V4L2_PIX_FMT_MJPEG;
        fmt.fmt.pix.field = V4L2_FIELD_INTERLACED;
        if (ioctl(fd_, VIDIOC_S_FMT, &fmt) == -1)
        {
            ROS_ERROR("Failed to set V4L2 format");
            close(fd_);
            return;
        }

        struct v4l2_requestbuffers req;
        req.count = 4;
        req.type = V4L2_BUF_TYPE_VIDEO_CAPTURE;
        req.memory = V4L2_MEMORY_MMAP;
        if (ioctl(fd_, VIDIOC_REQBUFS, &req) == -1)
        {
            ROS_ERROR("Failed to request buffers");
[... 210 lines cut ...]
    ros::Publisher image_pub_;
};

int main(int argc, char** argv)
{
    ros::init(argc, argv, "v4l2_camera_node");
    V4L2CameraNode node;
    ros::spin();
    return 0;
}

报错：
(base) orangepi@orangepi5plus:~/hk/code/v4l2$ rosrun v4l2 publisher 
[ INFO] [1728982422.451431991]: mpp create ret: 0
[ INFO] [1728982422.459096268]: mpp init  ret: 0
[ INFO] [1728982422.470774681]: BUffer[0] mapped at address: 0x7f89ec0000, length: 614400
[ INFO] [1728982422.724946479]: buf.bytesused: 63660
[ INFO] [1728982422.727768302]: Dequeued buffer: index=0, bytesused=63660, start address=0x7f89ec0000
[ INFO] [1728982422.728311668]: MPP packet initialized successfully
[ INFO] [1728982422.730474920]: Putting packet into MPP for decoding with size:63660
mpp_ctx_0x5571f6c2c0
mpp_pkt_0x5571f7b4a8
[ INFO] [1728982422.731062034]: MPP packet data ptr: 0x7f89ec0000, size: 63660
[ERROR] [1728982422.731212823]: Failed to put packet into MPP, error code: -1012
有大佬知道怎么回事吗？

Pale166, 2025-04-27:
请问你解决了吗？

byg-git, 2025-05-23:
楼主解决了吗？