rockchip-linux/mpp #702: 在RK3588上使用MPP1.0.5,使用mpi_enc_test编码yuv文件生成的h264文件色彩丢失加色彩异常，再用mpi_dec_test解码回去颜色恢复

mrjcf, 2024-11-01:
遇到两种情况：
1：使用mpi_dec_test解码纯红色的h264文件生成的yuv色彩混乱，再使用mpi_enc_test把生成的yuv编码回去色彩恢复。

[image]

2：使用mpi_enc_test把yuv文件编码为h264后，画面色彩丢失，再使用mpi_dec_test解码为yuv色彩恢复。其中原始yuv文件色彩正常，使用ffmpeg编码h264色彩也正常。
[image]
哪位大佬能解答一下这个情况，万分感谢

HermanChen, 2024-11-04:
注意下图像格式和 stride 问题

mrjcf, 2024-11-04:
> 注意下图像格式和 stride 问题
感谢回复，请问是注意那个图像的格式，是yuv的还是h264的，以及stride问题具体是那个方面，小弟生性愚钝，有望道兄解惑。

HermanChen, 2024-11-04:
编解码器输出的都是 YUV420SP 格式，也就是 NV12 格式的，注意下
stride 是指两行像素间的 byte 距离，和一行的有效像素数是不一样的，可以搜下相关概念

mrjcf, 2024-11-04:
> 编解码器输出的都是 YUV420SP 格式，也就是 NV12 格式的，注意下 stride 是指两行像素间的 byte 距离，和一行的有效像素数是不一样的，可以搜下相关概念
好的，谢谢道友解答，我去查阅相关信息，再遇到未知问题，还望道友解惑，小弟诚惶诚恐，不胜感激。

mrjcf, 2024-11-28:
> 你好，我也遇到了相同的问题，请问现在解决了吗

注意编码前后视频数据的格式