rockchip-linux/mpp #705: 为什么使用了MPP总是有内存无法释放

LeoD218, 2024-11-04:
释放MPP的代码如下
`//释放MPP解码器
bool MppDecode::mppDecodeFree()
{
    mApi->reset(mCtx);
    quit();
    wait();

    if (pktBuf)
    {
        mpp_buffer_put(pktBuf);
        pktBuf = NULL;
    }

    if (frmBuf)
    {
        mpp_buffer_put(frmBuf);
        frmBuf = NULL;
    }

    if (packet)
    {
        mpp_packet_deinit(&packet);
        packet = NULL;
    }

    if (frame)
    {
        mpp_frame_deinit(&frame);
        frame = NULL;
    }
    if(mCtx)
    {
        mpp_destroy(mCtx);
        mCtx = NULL;
    }

    if (frmGrp)
    {
        mpp_buffer_group_put(frmGrp);
        frmGrp = NULL;
    }

    if (pktGrp)
    {
        mpp_buffer_group_put(pktGrp);
        pktGrp = NULL;
    }

    if(dataBuf)
    {
        munmap(dataBuf,length);
        dataBuf = NULL;
    }
    // //释放检查
    // if (pktBuf == nullptr)
    // {
    //     qDebug()<<"pktBuf";
    // }
    // if (frmBuf == nullptr)
    // {
    //     qDebug()<<"frmBuf";
    // }
    // if (packet == nullptr)
    // {
    //     qDebug()<<"packet";
    // }
    // if (frame == nullptr)
    // {
    //     qDebug()<<"frame";
    // }
    // if(mCtx == nullptr)
    // {
    //     qDebug()<<"mCtx";
    // }
    // if (frmGrp == nullptr)
    // {
    //     qDebug()<<"frmGrp";
    // }
    // if (pktGrp == nullptr)
    // {
    //     qDebug()<<"pktGrp";
    // }
    // if(dataBuf == nullptr)
    // {
    //     qDebug()<<"databuf";
    // }
    return true;
}`
总是无法释放干净，希望能得到官方的答复

leashi, 2024-11-26:
有没有途径不用mpp？

HermanChen, 2024-11-26:
把 mpp_buffer 和 mpp_mem 里的调试 log 开关打开，分析 log

LeoD218, 2024-12-11:
> 把 mpp_buffer 和 mpp_mem 里的调试 log 开关打开，分析 log

请问要如何打开

Unlicensed-driver-ljx, 2024-12-27:
如何打开呢