rockchip-linux/mpp #707: eglCreateImageKHR 返回egl_bad_alloc（12291 ），有解决办法吗？

zhouxiaojun2008, 2024-11-06:
使用rk3588编码，输入原始图像通过qt OpenGLFramebufferObject::handle()拷贝过来的，在创建目的FBO的时候使用了egl的接口
`static PFNGLEGLIMAGETARGETTEXTURE2DOESPROC s_imageTarget2DFunc = nullptr;
            static PFNEGLCREATEIMAGEKHRPROC s_createImageFunc = nullptr;
            static PFNEGLDESTROYIMAGEKHRPROC s_destroyImageFunc = nullptr;
            if (s_createImageFunc == nullptr)
                s_createImageFunc = (PFNEGLCREATEIMAGEKHRPROC)eglGetProcAddress("eglCreateImageKHR");
            
            if (s_imageTarget2DFunc == nullptr)
                s_imageTarget2DFunc = (PFNGLEGLIMAGETARGETTEXTURE2DOESPROC)eglGetProcAddress("glEGLImageTargetTexture2DOES");
 
            if (s_destroyImageFunc == nullptr)
                s_destroyImageFunc = (PFNEGLDESTROYIMAGEKHRPROC)eglGetProcAddress("eglDestroyImageKHR");
            EGLImageKHR image;
            /* Cursor format should be ARGB8888 */
            const EGLint attrs[] = {
                EGL_IMAGE_PRESERVED_KHR, EGL_TRUE,
                EGL_WIDTH, m_frameWidth,
                EGL_HEIGHT, m_frameHeight,
                EGL_LINUX_DRM_FOURCC_EXT, DRM_FORMAT_RGB888,
                EGL_DMA_BUF_PLANE0_FD_EXT, m_frameBuffer->getFd(),
                EGL_DMA_BUF_PLANE0_OFFSET_EXT, 0,
                EGL_DMA_BUF_PLANE0_PITCH_EXT, m_frameWidth * 3,
                EGL_NONE,
            };

            EGLDisplay display = eglGetCurrentDisplay();
            glBindTexture(GL_TEXTURE_EXTERNAL_OES, m_texture);
            image = s_createImageFunc(display, EGL_NO_CONTEXT,
                                EGL_LINUX_DMA_BUF_EXT, NULL, attrs);
            if (image == EGL_NO_IMAGE) 
            {
                GLint maxTextureSize;
                glGetIntegerv(GL_MAX_TEXTURE_SIZE, &maxTextureSize);
                LOGE("create EGLImageKHR failed, size:{}x{}, maxTextureSize:{}, error {}", m_frameWidth, m_frameHeight, maxTextureSize, eglGetError());
                return false;
            }
            
            s_imageTarget2DFunc(GL_TEXTURE_EXTERNAL_OES, (GLeglImageOES)image);
            s_destroyImageFunc(display, image);`
当width和height过大的时候，就会报错比如传入宽高是2880*440是正常的，传入如果是3000*458 就会报错如下 create EGLImageKHR failed, maxTextureSize:16383, error 12291。

HermanChen, 2024-11-06:
这个我也不了解啊……

closed 2024-11-06 by HermanChen

zhouxiaojun2008, 2024-11-07:
这个是不是说明显存不够啊 ，是不是可以通过更改配置文件来给解码器，GPU分配更多的内存？
 如果是，怎么修改？我们有一些几个摄像头拼接的画面，尺寸会很大，现在尺寸超过3000就会eglCreateImageKHR调用就会失败说显存不足