rockchip-linux/mpp #709: MPP decoding artifacts when MPP_DEC_SET_PARSER_SPLIT_MODE is disabled for h265 intra-refresh video stream

gehee, 2024-11-08:
I am trying to decode an h265 video stream and decoding latency is critical in my application.

Without intra refresh enabled, my rk3588 is able to deocde the stream in ~1ms(4ms max) with the following config:
MPP_DEC_SET_PARSER_SPLIT_MODE: false;
MPP_DEC_SET_DISABLE_ERROR: true;
mpp_dec_cfg_set_u32(cfg, "base:split_parse", 1);

When I enable intra-refresh in my source stream to even out the packet size, mpp does not decode the stream properly with the same settings, it shows artifacts. 
To remove these artifacts, I need to set MPP_DEC_SET_PARSER_SPLIT_MODE = true but it increases latency to 9ms average (40ms max). 

Is this a bug ?
Is there anyother way to decode intrarefresh h265 stream while keeping decoding latency less than 1ms ?

HermanChen, 2024-11-11:
MPP_DEC_SET_PARSER_SPLIT_MODE = true will let mpp to extract the whole frame packet as decoder's input.
It means the spliter will not put a packet until next frame's packet comes to spliter which increases the latency.
I think MPP_DEC_SET_DISABLE_ERROR is enough for the intra fresh stream decoding.

gehee, 2024-11-11:
with MPP_DEC_SET_PARSER_SPLIT_MODE: false, MPP_DEC_SET_DISABLE_ERROR: true , I see artifacts with intrarefresh though so it does not seem to be enough.

gehee, 2024-11-19:
Here is a file with intrarefresh enabled: let me know if you can reproduce the artifacts.[
[intrarefresh_4M.zip](https://github.com/user-attachments/files/17809359/intrarefresh_4M.zip)
](url)

Looking at it with https://mdakram.com/media-parser-gui/#/hevc, I am seeing very small frame at the beginning of every GOP. I am wondering if that's what showing up as artifact on screen every second.

gehee, 2024-11-19:
I figured it out; it was a problem with my encoder, I didn't enough refresh line to refresh the whole picture.

closed 2024-11-19 by gehee