rockchip-linux/mpp #710: RV1126上无法调用rga进行编码前处理

shi156, 2024-11-09:
1.目前使用的是librga1.10.1，并使用预编译的静态库，是否存在版本差异导致驱动无法正常打开
2.dmesg报错：rga2: unknown ioctl cmd!
3.项目报错：
Failed to import virtualaddr for src channel! Invalid parameters: RGA_IOC_IMPORT_BUFFER fail! Invalid argument
Failed to import virtualaddr for dst channel! Invalid parameters: No error message, it has been cleared.
src handle: (nil), phy_addr: (nil), vir_addr: (nil), fd: 0
dst handle: (nil), phy_addr: (nil), vir_addr: (nil), fd: 0
14783 im2d_rga_impl rga_set_buffer_info(482): Invaild src buffer, no address available in src buffer, phy_addr = 0, fd = 0, vir_addr = 0, handle = 0
350, imresize failed! Invalid parameters: Invaild src buffer, no address available in src buffer, phy_addr = 0, fd = 0, vir_addr = 0, handle = 0
4.报错函数：
src.handle =  importbuffer_virtualaddr(src_buf_char, src_buf_size);
ret = imresize(src, dst);
所以想请问大佬在mpp编码前是否可以进行rga相关的前处理呢，试过rkmedia的方法，但是手上的板子不支持isp，导致此路不通。感谢！！！

shi156, 2024-11-11:
通过查询，rv1126开发板上没有相关驱动信息，dmesg |grep rga
[    2.910028] rga2: Driver loaded successfully ver:3.2.56726
[    2.910464] rga2: Module initialized.
[   37.374276] rga2: unknown ioctl cmd!
[   37.374694] rga2: dma_buf_get fail fd[3]
[   37.374722] rga2: src:rga2_get_img_info fail
[   37.375016] rga2: RGA2 : DMA buf copy error

xiekanglin022, 2025-01-15:
请问你是怎么解决的？是重新编译内核了吗？