rockchip-linux/mpp #714: 请问mpp解码后如何获取到physicaladdr，然后使用importbuffer_physicaladdr？

iGwkang, 2024-11-15:
https://github.com/airockchip/librga/issues/93

mpp解码中的内存使用内部分配模式:
```c++
mpp_buffer_group_get_internal(&m_buffer_group, MPP_BUFFER_TYPE_DRM);
```
解码后，通过 `mpp_buffer_get_fd(mpp_frame_get_buffer(frame))`拿到`fd`
拿到fd之后，使用`importbuffer_fd`可以转换成功，1920x1080 这个耗时大概5ms
```c++
void rga_convert_with_fd(int src_fd, int src_size, int dst_fd, int dst_size, int width, int height)
{
    rga_buffer_t src = {};
    rga_buffer_t dst = {};
    im_rect src_rect = {};
    im_rect dst_rect = {};
    rga_buffer_handle_t src_handle, dst_handle;

    src_handle = importbuffer_fd(src_fd, src_size);
    dst_handle = importbuffer_fd(dst_fd, dst_size);
    if (src_handle == 0 || dst_handle == 0) {
        RUN_LOG_ERR("import drm fd error!");
        return;
    }

    src = wrapbuffer_handle(src_handle, width, height, RK_FORMAT_YCbCr_420_SP);
    dst = wrapbuffer_handle(dst_handle, width, height, RK_FORMAT_BGRA_8888);

    int ret = imcheck(src, dst, src_rect, dst_rect);
    if (IM_STATUS_NOERROR != ret)
    {
        RUN_LOG_ERR("%d, check error! %s", __LINE__, imStrError((IM_STATUS)ret));
    }

    IM_STATUS status = imcvtcolor(src, dst, src.format, dst.format);
    if (status != IM_STATUS_SUCCESS)
    {
        RUN_LOG_ERR("convert error %s", imStrError(status));
    }
}
```

我想尝试一下`importbuffer_physicaladdr`看看耗时多少，我要怎么样才能拿到mpp解码后那一帧数据的物理地址？我尝试过`drmPrimeFDToHandle`+`drm_buf_get_phy`，获取出来的地址是0

HermanChen, 2024-11-15:
mpp 里不使用物理地址，因为没有可移植性，最多是使用映射后的 iova 地址
现在一般的非 cma 的 buffer 也是使用散列表的方式组织的，无法获取到某一个物理地址

iGwkang, 2024-11-29:
@HermanChen 如果使用外部分配内存的方式是否能做到？

HermanChen, 2024-11-29:
外部的 buffer 用 EXT_DMA 的 group 来 import 的话，有可能是可以的，只要是内核的能 import 进来能得到可以用地址就可以

closed 2024-12-03 by iGwkang