rockchip-linux/mpp #715: CPU memcpy效率低，如何使用buffer cached模式？

xyl147033, 2024-11-15:
出现的问题和#433《mpp解码后数据拷贝占用CPU，是否有更好方式进行拷贝？》类似，
目前使用硬件平台：rk3588
mpp版本：release
视频渲染软件使用 Qt5.15+OPENGL ES 3.2，将解码后的数据拷贝到GPU，利用OPENGL中的转换矩阵对YUV420SP数据转换并渲染
问题：
在从mpp的buffer中memcpy到OPENGL的PBO中时（数据量大小：1920*1088*1.5），需要15-30ms，
看过[HermanChen](https://github.com/HermanChen)的大部分解答，但目前并不知道如何去使用buffer cached模式，在mpp_buffer.h中只寻找到宏定义MPP_BUFFER_FLAGS_CACHABLE，且没有develop中的mpp_buffer_sync_begin和mpp_buffer_sync_end函数，请问各位大神有没有相关demo可以参考？

HermanChen, 2024-11-15:
在创建 MppBufferGroup 的时候，使用 MPP_BUFFER_TYPE_DMA_HEAP | MPP_BUFFER_FLAGS_CACHABLE 这样的类型

xyl147033, 2024-11-15:
> 在创建 MppBufferGroup 的时候，使用 MPP_BUFFER_TYPE_DMA_HEAP | MPP_BUFFER_FLAGS_CACHABLE 这样的类型

好的，谢谢解答。还有两个小问题：1.可以使用MPP_BUFFER_TYPE_DRM吗？2.不使用mpp_buffer_sync是否会产生其他问题啊

HermanChen, 2024-11-18:
可以使用 DRM，但 drm 的内核 cache 类型支持不一定在所有的内核上都正常，有一些老内核的 sync 函数是有问题的