rockchip-linux/mpp #716: 安卓运行mpp_platform_test没有反应，日志打印也没有，但是同样机子的linux版本上跑通了mpi_dec_test，在安卓上跑的话会报错段错误

realzzq, 2024-11-16:
系统版本
:/data/local/tmp # cat /proc/version

Linux version 5.10.110+ (ys@x99) (Android (7284624, based on r416183b) clang version 12.0.5 (https://android.googlesource.com/toolchain/llvm-project c935d99d7cf2016289302412d708641d52d2f7ee), LLD 12.0.5 (/buildbot/src/android/llvm-toolchain/out/llvm-project/lld c935d99d7cf2016289302412d708641d52d2f7ee)) #78 SMP PREEMPT Sat Feb 25 17:41:07 CST 2023

运行案例
:/data/local/tmp # ./mpp_platform_test                                                                                                                                                                                                                                                                             
:/data/local/tmp #

realzzq, 2024-11-17:
现在程序正常运行了，但是没输出解码后的yuv数据 命令行如下:

LD_PRELOAD=./libmpp.so ./mpi_dec_test -i input.h264  -o output.yuv -n 8

日志如下：
11-17 20:04:17.237  6406  6406 I mpi_dec_utils: input file input.h264 size 36334163
11-17 20:04:17.237  6406  6406 I mpi_dec_utils: cmd parse result:
11-17 20:04:17.237  6406  6406 I mpi_dec_utils: input  file name: input.h264
11-17 20:04:17.237  6406  6406 I mpi_dec_utils: output file name: output.yuv
11-17 20:04:17.237  6406  6406 I mpi_dec_utils: width      :    0
11-17 20:04:17.237  6406  6406 I mpi_dec_utils: height     :    0
11-17 20:04:17.237  6406  6406 I mpi_dec_utils: type       :    7
11-17 20:04:17.237  6406  6406 I mpi_dec_utils: max frames :    8
11-17 20:04:17.237  6406  6406 I mpi_dec_test: mpi_dec_test start
11-17 20:04:17.434  6406  6406 I mpp_info: mpp version: unknown mpp version for missing VCS info
11-17 20:04:17.434  6406  6406 I mpi_dec_test: 0xb400007b8659b130 mpi_dec_test decoder test start w 0 h 0 type 7
11-17 20:04:17.435  6406  6406 E mpp_platform: client 12 driver is not ready!
11-17 20:04:17.492  6406  6411 I mpi_dec_test: 0xb400007b8659b130 decode_get_frame get info changed found
11-17 20:04:17.492  6406  6411 I mpi_dec_test: 0xb400007b8659b130 decoder require buffer w:h [1920:1080] stride [1920:1088] buf_size 4177920
11-17 20:04:17.501  6406  6411 I mpi_dec_test: 0xb400007b8659b130 decode get frame 0
11-17 20:04:17.542  6406  6411 I mpi_dec_test: 0xb400007b8659b130 decode get frame 1
11-17 20:04:17.569  6406  6411 I mpi_dec_test: 0xb400007b8659b130 decode get frame 2
11-17 20:04:17.579  6406  6411 I mpi_dec_test: 0xb400007b8659b130 decode get frame 3
11-17 20:04:17.588  6406  6411 I mpi_dec_test: 0xb400007b8659b130 decode get frame 4
11-17 20:04:17.614  6406  6411 I mpi_dec_test: 0xb400007b8659b130 decode get frame 5
11-17 20:04:17.639  6406  6411 I mpi_dec_test: 0xb400007b8659b130 decode get frame 6
11-17 20:04:17.650  6406  6411 I mpi_dec_test: 0xb400007b8659b130 decode get frame 7
11-17 20:04:17.659  6406  6411 I mpi_dec_test: decode 8 frames time 222 ms delay  64 ms fps 35.97
11-17 20:04:17.662  6406  6406 I mpi_dec_test: test success max memory 23.91 MB

closed 2024-11-17 by realzzq

HDL199411, 2025-03-03:
你好，这是什么问题，我也遇到了