rockchip-linux/mpp #721: ffmpeg解复用avs2(ts封装)后给mpp库，只解码了12帧报错

wzw88486969, 2024-11-25:
Nov 24 11:44:45 topeet mpp[37334]: avs2d_dpb: Assertion ph->doi >= 0 && ph->poi >= 0 failed at compute_frame_order_index:244
Nov 24 11:44:45 topeet mpp[37334]: mpi_dec_test: decoder_get_frame get err info:0 discard:1.
Nov 24 11:44:45 topeet mpp[37334]: mpi_dec_test: decode_get_frame get frame 10
Nov 24 11:44:45 topeet mpp[37334]: mpi_dec_test: decode_get_frame get frame 11
Nov 24 11:44:45 topeet mpp[37334]: mpi_dec_test: base_length = 4177920
Nov 24 11:44:45 topeet mpp[37334]: avs2d_api: avs2d_parse Parse stream failed!
Nov 24 11:44:45 topeet mpp[37334]: avs2d_dpb: Assertion ph->doi >= 0 && ph->poi >= 0 failed at compute_frame_order_index:244
Nov 24 11:44:45 topeet mpp[37334]: avs2d_api: avs2d_parse Parse stream failed!
Nov 24 11:44:45 topeet mpp[37334]: avs2d_dpb: Assertion ph->doi >= 0 && ph->poi >= 0 failed at compute_frame_order_index:244
Nov 24 11:44:45 topeet mpp[37334]: mpi_dec_test: decode_get_frame get frame 12
Nov 24 11:44:45 topeet mpp[37334]: mpi_dec_test: base_length = 4177920
Nov 24 11:44:45 topeet mpp[37334]: avs2d_api: avs2d_parse Parse stream failed!
Nov 24 11:44:45 topeet mpp[37334]: avs2d_dpb: Assertion ph->doi >= 0 && ph->poi >= 0 failed at compute_frame_order_index:244
Nov 24 11:44:45 topeet mpp[37334]: avs2d_api: avs2d_parse Parse stream failed!
Nov 24 11:44:45 topeet mpp[37334]: avs2d_dpb: Assertion ph->doi >= 0 && ph->poi >= 0 failed at compute_frame_order_index:244
Nov 24 11:44:45 topeet mpp[37334]: mpi_dec_test: decoder_get_frame get err info:0 discard:1.

[l_avs2.zip](https://github.com/user-attachments/files/17896691/l_avs2.zip)

wenyue7, 2024-11-30:
你好，确认一下是怎么抽的码流？

Choise-ieee, 2025-01-05:
您好 您解决了么 我好像也遇到了同样的问题 av_read_frame后的es包给过去