rockchip-linux/mpp #726: mpp轮询开关多路解码，遇到mpp库段错误

mojiaj, 2024-12-02:
我做了一个应用，5s循环开启30路1080P30解码（具体操作为初始化mppdec，开始解码，反初始化mppdec）
该程序运行一段时间之后，会产生段错误

HermanChen, 2024-12-03:
fd 731 感觉 buffer 用得比较多 mpp_buffer_mmap  失败有可能是 4G 虚存都用光了？

mojiaj, 2024-12-03:
好的，谢谢，我检查一下虚拟内存的使用，但是有一次我看到的现象好像是返回-22，这个也是虚拟内存不够的问题吗
 mpp_serivce: mpp_service_ioc_attach_fd failed ret -1 errno 22 Invalid argument
Dec  3 07:06:45 user.err mpp[20731]: mpp_dma_heap: mmap failed: Bad file descriptor
Dec  3 07:06:45 user.err mpp[20731]: mpp_buffer: mpp_buffer_mmap buffer 0 group 7433 fd 444 map failed caller try_proc_dec_task
Dec  3 07:06:45 user.err mpp[20731]: mpp_buffer: Assertion dst != __null failed at mpp_buffer_write_with_caller:160
Dec  3 07:06:45 user.err mpp[20731]: mpp_log: mpp_dmabuf_sync_end ioctl failed for Bad file descriptor from try_proc_dec_task
Dec  3 07:06:45 user.err mpp[20731]: mpp_dma_heap: mmap failed: Invalid argument
Dec  3 07:06:45 user.err mpp[20731]: mpp_buffer: mpp_buffer_mmap buffer 0 group 7433 fd 444 map failed caller hal_h265d_vdpu34x_gen_regs
Dec  3 07:06:45 user.err mpp[20731]: mpp_buffer: Assertion p->info.ptr != __null failed at mpp_buffer_get_ptr_with_caller:178
Dec  3 07:06:45 user.err mpp[20731]: mpp_buffer: mpp_buffer_get_ptr buffer 0x7e30012288 ret NULL from hal_h265d_vdpu34x_gen_regs
Dec  3 07:06:45 kern.err kernel: [12861.149128] mpp_dma_import_fd:210: dma_buf_get fd 444 failed(-9)
Dec  3 07:06:45 kern.err kernel: [12861.149136] rk_vcodec: mpp_process_request:1374: can not import fd 444
Dec  3 07:06:45 kern.err kernel: [12861.149139] rk_vcodec: mpp_collect_msgs:1562: session 3535 process cmd 401 ret -22
Dec  3 07:06:45 kern.err kernel: [12861.149141] rk_vcodec: mpp_dev_ioctl:1672: collect msgs failed -22

HermanChen, 2024-12-03:
文件句柄不对？还是说内核比较老，不支持 attach 这个 ioctl？

mojiaj, 2024-12-03:
程序正常是可以跑起来一段时间的，内核是5.1版本的。如果应用程序有内存越界的话，也会影响这个的吗？这个好像是mpp内部申请的buffer

HermanChen, 2024-12-03:
应用程序有内存越界会有影响，传一份完整的从开始到出问题的 log，放到网盘或者其他地方

HermanChen, 2024-12-03:
另外，not permit 的话，用 root 跑下看看

mojiaj, 2024-12-03:
好的

mojiaj, 2024-12-04:
[var_log_messag.log](https://github.com/user-attachments/files/18001025/var_log_messag.log)
[串口.log](https://github.com/user-attachments/files/18001028/default.log)
0x0000007ff4a08018 in __memset_generic () from /lib/libc.so.6
(gdb) bt
#0  0x0000007ff4a08018 in __memset_generic () from /lib/libc.so.6
#1  0x0000007ff7e197a4 in hal_h265d_vdpu34x_gen_regs () from /lib/librockchip_mpp.so.1
#2  0x0000007ff7d67778 in mpp_dec_parser_thread(void*) () from /lib/librockchip_mpp.so.1
#3  0x0000007ff49f0958 in start_thread () from /lib/libc.so.6
#4  0x0000007ff4a4609c in thread_start () from /lib/libc.so.6

HermanChen, 2024-12-04:
码流是有错的吧，解码器一段时间会报一下错，后面 410 fd 开始有问题，应该是上层哪里错关了 fd 导致的？

mojiaj, 2024-12-16:
码流是一直丢包导致的，问题找到啦，应用程序有地方越界了，非常感谢

closed 2024-12-16 by mojiaj