rockchip-linux/mpp #727: rk356x关于drm cursor格式的问题

iGwkang, 2024-12-04:
抱歉，这个问题卡了快一个月了，实在是在网上没有找到什么资料，所以来这里想问一下这个问题。
使用drm渲染时，鼠标光标无法显示，光标数据的格式是ARGB，但鼠标事件时能收到的
`drmModeSetCursor2` 调用鼠标相关的函数在dmesg中会报错：
> [10423.793739] [drm:vop2_plane_atomic_check] *ERROR* Unsupported linear format at Cluster0-win0
[10424.280492] [drm:vop2_plane_atomic_check] *ERROR* Unsupported linear format at Cluster0-win0
[10424.361392] [drm:vop2_plane_atomic_check] *ERROR* Unsupported linear format at Cluster0-win0
[10424.476362] [drm:vop2_plane_atomic_check] *ERROR* Unsupported linear format at Cluster0-win0
[10424.525402] [drm:vop2_plane_atomic_check] *ERROR* Unsupported linear format at Cluster0-win0
[10426.419508] [drm:vop2_plane_atomic_check] *ERROR* Unsupported linear format at Cluster0-win0
[10426.446422] [drm:vop2_plane_atomic_check] *ERROR* Unsupported linear format at Cluster0-win0
[10426.463126] [drm:vop2_plane_atomic_check] *ERROR* Unsupported linear format at Cluster0-win0
[10426.501060] [drm:vop2_plane_atomic_check] *ERROR* Unsupported linear format at Cluster0-win0

然后我又看到一些资料上写3566/3568的Cluster图层只支持`AFBC`格式，但是`AFBC`也没有什么资料。
还有就是运行Qt程序也是没有鼠标，Qt使用`eglfs`和`eglfs_kms`渲染
```
export QT_QPA_PLATFORM=eglfs
export QT_QPA_EGLFS_INTEGRATION=eglfs_kms
```
程序启动之后 移动一下鼠标也会报上面的错
Qt要指定`QT_QPA_EGLFS_KMS_CONFIG`配置文件，将`hwcursor`设置成false才会显示鼠标，但是这样显示出来的鼠标非常卡顿，这个应该是Qt自己画出来的鼠标。

我最终想问的是rk3566/3568的`Cluster`图层能否支持ARGB之类的格式？
如果能支持是否有示例参考？
假如无法支持ARGB，我应该怎么样才能正常把鼠标光标显示出来，是否有ARGB转AFBC之类的API，即使能够转换，但Qt还是无法正常显示，因为Qt的源码已经定死了，我是否要改Qt源码才能实现？

Jerzha, 2024-12-05:
AFBC不是必选项，只支持AFBC指的是他只支持 Arm格式的FBC，不支持其他格式FBC，而不是只支持FBC。几乎所有VOP图层都是支持ARGB的，参考示例参看网络上DRM VOP图层合并代码，是标准的atomic api用法。QT我们官方是不支持的，请自行联系QT FAE支持。

iGwkang, 2024-12-05:
@Jerzha 感谢回复
我先看一下vop图层合并。但是我还是很疑惑，为何在3328/3528平台没有这些问题？

iGwkang, 2024-12-05:
@Jerzha 我使用drm接口往cursor图层添加数据时，依旧会报错
```
uint32_t fb_id = 0;
uint32_t handles[4] = { 0x0 };
uint32_t pitches[4] = { 0x0 };
uint32_t offsets[4] = { 0x0 };
handles[0] = m_cursor_dumb_handle;
pitches[0] = 4 * 32;

if (0 != drmModeAddFB2(m_drm_fd, 32, 32, DRM_FORMAT_ARGB8888, handles, pitches, offsets, &fb_id, 0))
{
    printf("add cursor fb failed");
}

int ret = drmModeSetPlane(m_drm_fd, 71, 85, fb_id, 0, 100, 100, 32, 32, 0, 0, 32 << 16, 32 << 16);
if (ret != 0)
{
    printf("set cursor plane failed, ret=%d, err_str=%s", ret, strerror(errno));
}
```

程序输出：
```
set cursor plane failed, ret=-22, err_str=Invalid argument
```
dmesg输出：
```
[10758.977304] [drm:vop2_plane_atomic_check] *ERROR* Unsupported linear format at Cluster0-win0
```

Jerzha, 2024-12-11:
你这段代码里问题一堆，就一个plane为什么handle和pitch给了4个数组，pitch的计算也有问题，你的宽度是只有32吗？drmModeSetPlane的参数都是哪里来的？planeid和crtcid怎么来的？src w和h为什么只有100？目标宽高确是32？
建议你先去看下libdrm里modetest代码参考下，或者参看rk的aosp中hwc的写法。
本问题和mpp无关，不再回复。

iGwkang, 2024-12-11:
@Jerzha 
> planeid和crtcid怎么来的？src w和h为什么只有100？目标宽高确是32？

planeid和crtcid是系统日志里的，这是示例代码没有写获取id的过程，直接从/sys/kernel/debug/dri/0/state拿到的，两个100是x和y，是屏幕坐标。

> 建议你先去看下libdrm里modetest代码参考下，或者参看rk的aosp中hwc的写法。

可能你没有试过rk3566的modetest吧，你试过就知道了，使用-C参数测试鼠标，报一样的错，drm最新的代码也是一样的。rk3528上是正常的，rk给我的回复是3528使用的VOP1.0架构。3566使用的是VOP2.0架构，总之2.0需要转换，破坏了兼容性。

我的问题已经找到了解决办法。

closed 2024-12-11 by iGwkang

muzhiyun, 2025-09-15:
@iGwkang  不好意思，能请教下如何解决的吗？

我目前在RK3568平台同样也遇到送RGBA给这个图层报同样错误，也看到pdf中声明Cluster0图层有注明只支持压缩数据不支持非压缩数据，有尝试改为其他图层(例如Esmart0或Smart0，仅调整Volayer其余代码保持不动)就能变为正常播放。所以现在也想问同样的问题“是否有ARGB转AFBC之类的API？”