rockchip-linux/mpp #730: rc: Assertion rate_in >= rate_out failed at rc_frm_check_drop:154

denghengli, 2024-12-05:
1、软件版本：
mpp：1.0.7
kernel：5.10

2、应用说明
我的应用是这样的：解码rtsp码流解码后再编码存成起来，摄像头帧率是25fps，开了两个线程分别视频跟图片，处理流程如下：
（1）视频编码线程： rtsp取流 --> mpp解码 --> mpp编码+字符叠加 --> 如果是一个 I 帧则发送到另一个图片编码线程中
（2）图片编码线程：mpp解码（视频编码线程中mpp编码生成的I帧）--> mjpeg编码生成图片

3、问题描述
如果只有视频编码线程的话运行正常，但是如果有图片处理线程，每次在编码图片的时候就会报错
rc: Assertion rate_in >= rate_out failed at rc_frm_check_drop:154
内核打印如下信息：
[   68.902380] rk_iommu fdee0800.iommu: Page fault at 0x00000000ff99d000 of type read
[   68.902453] rk_iommu fdee0800.iommu: iova = 0x00000000ff99d000: dte_index: 0x3fe pte_index: 0x19d page_offset: 0x0
[   68.902468] rk_iommu fdee0800.iommu: mmu_dte_addr: 0x0000000000281000 dte@0x0000000000281ff8: 0x2608001 valid: 1 pte@0x0000000002608674: 0x000000 valid: 0 page@0x0000000000000000 flags: 0x0
[   68.902485] rk_iommu fdee0800.iommu: fault addr 0xff99d000 status 4b arg 00000000504a0872
[   68.902498] rk_vcodec: mpp_task_dump_mem_region:2003: --- dump task 13 mem region ---
[   68.902507] rk_vcodec: mpp_task_dump_mem_region:2008: reg[ 48]: 0x00000000ff6a0000, size 2fd000
[   68.902514] rk_vcodec: mpp_task_dump_mem_region:2008: reg[ 49]: 0x00000000ff6a0000, size 2fd000
[   68.902522] rk_vcodec: mpp_task_dump_mem_region:2008: reg[ 50]: 0x00000000ff6a0000, size 2fd000
[   68.902529] rk_vcodec: mpp_task_dump_mem_region:2008: reg[ 77]: 0x00000000ff370000, size 32a000
[   68.902537] rk_vcodec: mpp_task_dump_hw_reg:2049: --- dump hardware register ---
[   68.902545] rk_vcodec: mpp_task_dump_hw_reg:2053: reg[000]: 0000: 0x00000000
[   68.902552] rk_vcodec: mpp_task_dump_hw_reg:2053: reg[001]: 0004: 0x00000000
[   68.902560] rk_vcodec: mpp_task_dump_hw_reg:2053: reg[002]: 0008: 0x070a141d
[   68.902567] rk_vcodec: mpp_task_dump_hw_reg:2053: reg[003]: 000c: 0x090e1a25
[   68.902574] rk_vcodec: mpp_task_dump_hw_reg:2053: reg[004]: 0010: 0x04060609
[   68.902582] rk_vcodec: mpp_task_dump_hw_reg:2053: reg[005]: 0014: 0x06080a0c
[   68.902589] rk_vcodec: mpp_task_dump_hw_reg:2053: reg[006]: 0018: 0x0f161f26
[   68.902597] rk_vcodec: mpp_task_dump_hw_reg:2053: reg[007]: 001c: 0x161a2327
[   68.902603] rk_vcodec: mpp_task_dump_hw_reg:2053: reg[008]: 0020: 0x0a0a1014
[   68.902610] rk_vcodec: mpp_task_dump_hw_reg:2053: reg[009]: 0024: 0x10171723
[   68.902617] rk_vcodec: mpp_task_dump_hw_reg:2053: reg[010]: 0028: 0x1b20292d
[   68.902625] rk_vcodec: mpp_task_dump_hw_reg:2053: reg[011]: 002c: 0x2c2a3028
[   68.902632] rk_vcodec: mpp_task_dump_hw_reg:2053: reg[012]: 0030: 0x14181c20
[   68.902639] rk_vcodec: mpp_task_dump_hw_reg:2053: reg[013]: 0034: 0x18161619
[   68.902645] rk_vcodec: mpp_task_dump_hw_reg:2053: reg[014]: 0038: 0x292d3029
[   68.902653] rk_vcodec: mpp_task_dump_hw_reg:2053: reg[015]: 003c: 0x1f252828
[   68.902660] rk_vcodec: mpp_task_dump_hw_reg:2053: reg[016]: 0040: 0x07070a13
[   68.902667] rk_vcodec: mpp_task_dump_hw_reg:2053: reg[017]: 0044: 0x07080a1a
[   68.902674] rk_vcodec: mpp_task_dump_hw_reg:2053: reg[018]: 0048: 0x28282828
[   68.902682] rk_vcodec: mpp_task_dump_hw_reg:2053: reg[019]: 004c: 0x28282828
[   68.902689] rk_vcodec: mpp_task_dump_hw_reg:2053: reg[020]: 0050: 0x0a0a1628
[   68.902696] rk_vcodec: mpp_task_dump_hw_reg:2053: reg[021]: 0054: 0x131a2828
[   68.902703] rk_vcodec: mpp_task_dump_hw_reg:2053: reg[022]: 0058: 0x28282828
[   68.902711] rk_vcodec: mpp_task_dump_hw_reg:2053: reg[023]: 005c: 0x28282828
[   68.902718] rk_vcodec: mpp_task_dump_hw_reg:2053: reg[024]: 0060: 0x28282828

**但是编码生成的视频跟图片都正常（以为不是什么大问题所以没管），但是运行了一段时间后编码器就异常了，隐患非常大**

4、问题
rc: Assertion rate_in >= rate_out failed at rc_frm_check_drop:154 这个报错什么意思呢？ 什么原因会导致这个
可以从哪些方面如果排查或是优化呢

HermanChen, 2024-12-06:
ff6a0000 + 2fd000 = ff99d000
可以看到是编码的输入图像访问越界了，检查下输入 buffer 的配置

rc: Assertion rate_in >= rate_out failed at rc_frm_check_drop:154
这个是指输入帧率不能小于输出帧率，编码器不会增加帧出来

denghengli, 2024-12-06:
1、rc: Assertion rate_in >= rate_out failed at rc_frm_check_drop:154 这个确实是我配置得有问题，fps_in_num为0了，改了确实没在报错了
    /* fix input / output frame rate */
    mpp_enc_cfg_set_s32(cfg, "rc:mode", pMppEncData->rc_mode);
    mpp_enc_cfg_set_s32(cfg, "rc:fps_in_flex", pMppEncData->fps_in_flex);
    mpp_enc_cfg_set_s32(cfg, "rc:fps_in_num", pMppEncData->fps_in_num);
    mpp_enc_cfg_set_s32(cfg, "rc:fps_in_denorm", pMppEncData->fps_in_den);
    mpp_enc_cfg_set_s32(cfg, "rc:fps_out_flex", pMppEncData->fps_out_flex);
    mpp_enc_cfg_set_s32(cfg, "rc:fps_out_num", pMppEncData->fps_out_num);
    mpp_enc_cfg_set_s32(cfg, "rc:fps_out_denorm", pMppEncData->fps_out_den);
    mpp_enc_cfg_set_s32(cfg, "rc:gop", pMppEncData->gop ? pMppEncData->gop : pMppEncData->fps_out_num * 2);

2、“ff6a0000 + 2fd000 = ff99d000，可以看到是编码的输入图像访问越界了，检查下输入 buffer 的配置”
这个是指检查哪个的输入buffer呢，什么条件会触发？ 
这个问题是偶发的，看结果我的视频跟图片编解码都正常，如果长时间这样运行会有什么问题吗