rockchip-linux/mpp #732: RK3588使用MPP进行多路同时编解码导致编码输出卡顿的问题

GUANJIAA, 2024-12-07:
MPP库解码器配置：内存分配采用纯内部分配模式，解码模式都配置了非阻塞式输入输出模式。
MPP库编码器配置：编码模式都配置了非阻塞式输入输出模式。

在RK3588的机器上，使用MPP库同时编解码16路的1080P的RTSP流，发现编码器要输出IDR帧时，调用encode_get_packet接口耗时会达到300ms。随着路数的减少，调用encode_get_packet接口耗时会减少。

编解码过程：
获取RTSP流的视频帧，调用decode_put_packet接口塞入解码器。
另外一条线程，调用decode_get_frame接口从解码器中获取YUV图像，再调用encode_put_frame接口塞入到编码器中，再调用encode_get_packet接口获取重编码后的视频帧

麻烦问下有什么方法可以减少encode_get_packet接口的耗时，或者是加快RK3588使用MPP库的重编码的速度吗。非常感谢

GUANJIAA, 2024-12-07:
MPP库的版本：1.0.5，1.0.7都会出现此现象

HermanChen, 2024-12-09:
硬件是时分复用的，接口延迟增加是正常的，但是耗时 300ms 不太正常，16 路的 1080p，每路都编码一帧也不到 60ms
echo 0x100 > /sys/module/rk_vcodec/parameters/mpp_dev_debug （手打，大概这样）
看下硬件时间

GUANJIAA, 2024-12-09:
[image]
是不是可以这样算硬件时间，每个VPU有2个编码核心和2个解码核心，16路1080P，每个任务编一帧的耗时为3.8ms，16路的话，每路的编一帧时间大概在30ms左右

HermanChen, 2024-12-09:
16路的时间估计是这样的，应该不会超出太多才是
1. I 帧耗时 300ms 这不太正常，硬件时间正常么？
2. 这不太对，看下对应硬件时间和软件接口上看到的时间，是不是哪里有问题

GUANJIAA, 2024-12-09:
我已经找到了问题了，是因为编码器的配置不合理，在初始化时，编码器的h264:log2_max_frm_num配置参数的值赋值0了。将其赋值为1时，在编码I帧时，耗时就在30ms。谢谢你的解决方法的思路。

closed 2024-12-10 by HermanChen

Liaoyyyy, 2024-12-10:
你好，关于h264:log2_max_frm_num配置参数导致编码耗时异常的问题，我们尝试本地复现没有成功，测试中软硬件编码耗时都是正常的，是不是还有其他参数导致的这个耗时异常？

Liaoyyyy, 2024-12-11:
[0001-draft-enc_time-get-out_put-time-when-enc.patch](https://github.com/user-attachments/files/18092261/0001-draft-enc_time-get-out_put-time-when-enc.patch)
参数没看出明显问题，辛苦打上这个log补丁重新编译mpp库后复现一遍，然后log发出来看下。
测试时带上
echo 0x100 > /sys/module/rk_vcodec/parameters/mpp_dev_debug
一起对硬件时间进行统计