rockchip-linux/mpp #739: 使用OpenCV VideoCapture打开USB相机后初始化MPP异常

KiosWong, 2024-12-15:
使用OpenCV VideoCapture打开USB相机后初始化MPP异常,必须调用VideoCapture::release关闭相机后才可正常初始化MPP.

KiosWong, 2024-12-15:
``//------------------------------------------------------------------------------
//功能：初始化MPP编码器
//------------------------------------------------------------------------------
bool mpp_encode_init(MpiEncLoopData *data, MpiEncCmd *cmd, char **SPS_buf, int *SPS_length)
{
    MPP_RET ret;
    MppCtx ctx          = NULL;
    MppApi *mpi         = NULL;
    MppPacket packet    = NULL;
    MppFrame  frame     = NULL;

    // encode codec
    MppEncCodecCfg codec_cfg;
    memset(&codec_cfg, 0, sizeof(MppEncCodecCfg));
    MppEncPrepCfg prep_cfg;
    memset(&prep_cfg, 0, sizeof(MppEncPrepCfg));
    MppEncRcCfg rc_cfg;
    memset(&rc_cfg, 0, sizeof(MppEncRcCfg));

    
    //0--get paramter from cmd
    data->width         = cmd->width;
    data->height        = cmd->height;
    data->hor_stride    = MPP_ALIGN(cmd->width, 16);
    data->ver_stride    = MPP_ALIGN(cmd->height, 16);
    data->fmt           = cmd->format;
    data->type          = cmd->type;
    data->num_frames    = cmd->num_frames;
    data->frame_size    = data->hor_stride * data->ver_stride * 3 / 2;
    data->packet_size   = data->width * data->height;
    data->fps           = cmd->fps;
    data->gop           = cmd->gop;
    data->bps           = data->width * data->height / 10 * data->fps;

    printf("data->frame_size = %d----------------\n", data->frame_size);
    ret = mpp_buffer_get(NULL, &data->frm_buf, data->frame_size);
    if (ret)
    {
        printf("failed to get buffer for input frame ret %d\n", ret);
        return false;
    }
    //
    printf("mpi_enc_test encoder test start: w %d h %d type %d\n",data->width, data->height, data->type);
    //encoder demo
    ret = mpp_create(&ctx, &mpi);
    if (ret)
    {
        printf("mpp_create failed ret %d\n", ret);
        return false;
    }
    //
    ret = mpp_init(ctx, MPP_CTX_ENC, data->type);
    if (ret)
    {
        printf("mpp_init failed ret %d\n", ret);
        mpp_encode_userspace_deinit(ctx, data);
    }
    //
    //1--输入控制配置
    prep_cfg.change        = MPP_ENC_PREP_CFG_CHANGE_INPUT | MPP_ENC_PREP_CFG_CHANGE_ROTATION | MPP_ENC_PREP_CFG_CHANGE_FORMAT;
    prep_cfg.width         = data->width;
    prep_cfg.height        = data->height;
    prep_cfg.hor_stride    = data->hor_stride;
    prep_cfg.ver_stride    = data->ver_stride;
    if (ret)
    {
        printf("mpi control enc set prep cfg failed ret %d\n", ret);
        mpp_encode_userspace_deinit(ctx, data);
        return false;
    }
[... 103 lines cut ...]
        }
        // get and write sps/pps for H.264
        if (packet)
        {
            void *ptr    = mpp_packet_get_pos(packet);
            size_t len    = mpp_packet_get_length(packet);
            *SPS_buf = new char[len];
            memcpy(*SPS_buf, ptr, len);
            *SPS_length = len;
            printf("SPS_len = %d.\r\n", len);
            packet = NULL;
        } else {

        }
    }

    //get paramter from cmd
    data->ctx = ctx;
    data->mpi = mpi;
    data->prep_cfg = prep_cfg;
    data->codec_cfg = codec_cfg;
    data->rc_cfg = rc_cfg;

    return true;
}

HermanChen, 2024-12-16:
这套的结构体配置接口已经不再使用了，换新的字符串配置接口兼容性会比较好