rockchip-linux/mpp #740: rk3588 对进行视频文件进行H264编码，显示不能完全进行编码，本地ffplay编码后的视频播放速度很快

suyian7, 2024-12-16:
rkmpp_encoder.cpp
`#include "rkmpp_encoder.h"

RkEncoder::RkEncoder(Encoder_Param_t param):m_param_(param)
{
    // m_param_.hor_stride = m_param_.width;
	// m_param_.ver_stride = m_param_.height;
    m_param_.hor_stride = MPP_ALIGN(m_param_.width, 16);
    m_param_.ver_stride = MPP_ALIGN(m_param_.height, 16);

    //set frame_size
    if (m_param_.fmt == MPP_FMT_YUV420P){
        m_param_.frame_size = m_param_.hor_stride * m_param_.ver_stride * 3 / 2;
    }
    else if (m_param_.fmt <= MPP_FMT_YUV422SP_VU){
        m_param_.frame_size = m_param_.hor_stride * m_param_.ver_stride * 2;
    }
    else if (m_param_.fmt <= MPP_FMT_YUV422_VYUY){
        m_param_.hor_stride *= 2;
        m_param_.frame_size = m_param_.hor_stride * m_param_.ver_stride;
    }
    else{
        m_param_.frame_size = m_param_.hor_stride * m_param_.ver_stride * 4;
    }

    std::cout << "frame_size" << m_param_.frame_size << std::endl;
}

RkEncoder::~RkEncoder()
{
    if(m_ctx_){
		m_mpi_->reset(m_ctx_);
		mpp_destroy(m_ctx_);
        m_ctx_ = nullptr;
	}
    if (m_buf_){
        mpp_buffer_put(m_buf_);
        m_buf_ = nullptr;
    }
    if (m_frame_){
        mpp_frame_deinit(&m_frame_);
    }
    if (m_pkt_){
        mpp_packet_deinit(&m_pkt_);
    }
}

void RkEncoder::init_encoder()
{
    MppEncCfg m_cfg_ = nullptr;
    MPP_RET ret = MPP_OK;

    ret = mpp_create(&m_ctx_, &m_mpi_);
    if(ret != MPP_OK){
        std::cerr << "Failed to create Mpp context.Error:" << ret << std::endl;
        return;
    }

    ret = mpp_init(m_ctx_, MPP_CTX_ENC, MPP_VIDEO_CodingAVC);
    if(ret != MPP_OK){
        std::cerr << "Failed to initialize Mpp encoder.Error:" << ret << std::endl;
        return;
    }
    
    ret = mpp_buffer_get(NULL, &m_buf_, m_param_.frame_size*3);
    if (ret != MPP_OK) {
        std::cerr << "Failed to allocate MPP buffer" << std::endl;
        return;
    }

[... 224 lines cut ...]
Encoded frame: 15/171
Encoded frame: 16/171
Encoded frame: 17/171
Encoded frame: 18/171
Encoded frame: 19/171
Encoded frame: 20/171
Encoded frame: 21/171
Encoded frame: 22/171
Encoded frame: 23/171
Encoded frame: 24/171
Encoded frame: 25/171
Encoded frame: 26/171
Encoded frame: 27/171
Encoded frame: 28/171
Encoded frame: 29/171
Encoded frame: 30/171
Encoded frame: 31/171
Encoded frame: 32/171
Encoded frame: 33/171
Encoded frame: 34/171
Encoded frame: 35/171
Encoded frame: 36/171
Encoding completed. Output saved to output.h264 0

请问是哪里出了问题？

HermanChen, 2024-12-16:
先用 mpi_enc_test 做下测试看看

suyian7, 2024-12-16:
> 先用 mpi_enc_test 做下测试看看

命令：sudo mpi_enc_test -i /home/ema/wo
rk/rkmpp_demo/rkmpp_encoder/video/oceans.mp4 -t 7 -n 1116 -o /home/ema/work/rkmpp_demo/rkmpp_encoder/video/encode.h264 -w 960 -h 400 -fps 23
运行日志：
> Dec 16 13:40:47 ema mpp[4129]: mpp_info: mpp version: ad359dac author: dengkx        2024-11-14 Fix[h264d_vdpu34x]: disable cabac err check for rk356x
Dec 16 13:40:47 ema mpp[4129]: mpp_dma_heap: os_allocator_dma_heap_open open dma heap type 0 system failed!
Dec 16 13:40:47 ema mpp[4129]: mpp_buffer: MppBufferService get_group failed to get allocater with mode 0 type 20001
Dec 16 13:40:47 ema mpp[4129]: mpp_buffer: mpp_buffer_group_limit_config input invalid group (nil)
Dec 16 13:40:47 ema mpp[4129]: mpp_platform: client 12 driver is not ready!
Dec 16 13:40:47 ema mpp[4129]: mpp_dma_heap: os_allocator_dma_heap_open open dma heap type 0 system-uncached failed!
Dec 16 13:40:47 ema mpp[4129]: mpp_buffer: MppBufferService get_group failed to get allocater with mode 0 type 1
Dec 16 13:40:47 ema mpp[4129]: mpp_hal: mpp_hal_init hal h264d_rkdec init failed ret -1
Dec 16 13:40:47 ema mpp[4129]: mpp_hal: mpp_hal_init could not found coding type 7
Dec 16 13:40:47 ema mpp[4129]: mpp_dec: mpp_dec_init could not init hal
Dec 16 13:40:47 ema mpp[4129]: mpp_time: mpp_clock_put invalid clock (nil)
Dec 16 13:40:47 ema mpp[4129]: message repeated 10 times: [ mpp_time: mpp_clock_put invalid clock (nil)]
Dec 16 13:40:47 ema mpp[4129]: mpp: error found on mpp initialization
Dec 16 13:40:47 ema mpp[4129]: mpp_info: mpp version: ad359dac author: dengkx        2024-11-14 Fix[h264d_vdpu34x]: disable cabac err check for rk356x
Dec 16 13:40:47 ema mpp[4129]: mpp_dma_heap: os_allocator_dma_heap_open open dma heap type 0 system failed!
Dec 16 13:40:47 ema mpp[4129]: mpp_buffer: MppBufferService get_group failed to get allocater with mode 0 type 20001
Dec 16 13:40:47 ema mpp[4129]: mpp_buffer: mpp_buffer_group_limit_config input invalid group (nil)
Dec 16 13:40:47 ema mpp[4129]: mpp_dma_heap: os_allocator_dma_heap_open open dma heap type 0 system-uncached failed!
Dec 16 13:40:47 ema mpp[4129]: mpp_buffer: MppBufferService get_group failed to get allocater with mode 0 type 1
Dec 16 13:40:47 ema mpp[4129]: mpp_hal: mpp_hal_init hal h264d_rkdec init failed ret -1
Dec 16 13:40:47 ema mpp[4129]: mpp_hal: mpp_hal_init could not found coding type 7
Dec 16 13:40:47 ema mpp[4129]: mpp_dec: mpp_dec_init could not init hal
Dec 16 13:40:47 ema mpp[4129]: mpp_time: mpp_clock_put invalid clock (nil)
Dec 16 13:40:47 ema mpp[4129]: message repeated 10 times: [ mpp_time: mpp_clock_put invalid clock (nil)]
Dec 16 13:40:47 ema mpp[4129]: mpp: error found on mpp initialization
Dec 16 13:40:47 ema mpp[4129]: mpp_info: mpp version: ad359dac author: dengkx        2024-11-14 Fix[h264d_vdpu34x]: disable cabac err check for rk356x
Dec 16 13:40:47 ema mpp[4129]: mpp_dma_heap: os_allocator_dma_heap_open open dma heap type 0 system-uncached failed!
Dec 16 13:40:47 ema mpp[4129]: mpp_buffer: MppBufferService get_group failed to get allocater with mode 0 type 1
Dec 16 13:40:47 ema mpp[4129]: mpp_dma_heap: os_allocator_dma_heap_open open dma heap type 0 system-uncached failed!
Dec 16 13:40:47 ema mpp[4129]: mpp_buffer: MppBufferService get_group failed to get allocater with mode 0 type 1
Dec 16 13:40:47 ema mpp[4129]: mpp_dma_heap: os_allocator_dma_heap_open open dma heap type 0 system-uncached failed!
Dec 16 13:40:47 ema mpp[4129]: mpp_buffer: MppBufferService get_group failed to get allocater with mode 0 type 1
Dec 16 13:40:47 ema mpp[4129]: hal_h264e_vepu580: hal_h264e_vepu580_init init vepu buffer failed ret: -1
Dec 16 13:40:47 ema mpp[4129]: mpp_enc_hal: mpp_enc_hal_init hal hal_h264e init failed ret -1
Dec 16 13:40:47 ema mpp[4129]: mpp_enc_hal: mpp_enc_hal_init could not found coding type 7
Dec 16 13:40:47 ema mpp[4129]: mpp_enc: mpp_enc_init_v2 could not init enc hal
Dec 16 13:40:47 ema mpp[4129]: mpp: error found on mpp initialization
Dec 16 13:40:47 ema mpp[4129]: mpp_buffer: mpp_buffer_get_ptr invalid NULL input from encode
Dec 16 13:41:40 ema mpp[4228]: mpi_enc_utils: cmd parse result:
Dec 16 13:41:40 ema mpp[4228]: mpi_enc_utils: input  file name: /home/ema/work/rkmpp_demo/rkmpp_encoder/video/oceans.mp4
Dec 16 13:41:40 ema mpp[4228]: mpi_enc_utils: output file name: /home/ema/work/rkmpp_demo/rkmpp_encoder/video/encode.h264
Dec 16 13:41:40 ema mpp[4228]: mpi_enc_utils: width      : 960
Dec 16 13:41:40 ema mpp[4228]: mpi_enc_utils: height     : 400
Dec 16 13:41:40 ema mpp[4228]: mpi_enc_utils: format     : 0
Dec 16 13:41:40 ema mpp[4228]: mpi_enc_utils: type       : 7
Dec 16 13:41:40 ema mpp[4228]: mpi_enc_test: mpi_enc_test start
Dec 16 13:41:40 ema mpp[4228]: mpp_info: mpp version: ad359dac author: dengkx        2024-11-14 Fix[h264d_vdpu34x]: disable cabac err check for rk356x
Dec 16 13:41:40 ema mpp[4228]: mpi_enc_test: 0x7fa8001960 encoder test start w 960 h 400 type 7
Dec 16 13:41:40 ema mpp[4228]: mpp_platform: client 12 driver is not ready!
Dec 16 13:41:40 ema mpp[4228]: mpp_enc: MPP_ENC_SET_RC_CFG bps 1104000 [69000 : 1173000] fps [23:23] gop 46
Dec 16 13:41:40 ema mpp[4228]: h264e_api_v2: MPP_ENC_SET_PREP_CFG w:h [960:400] stride [960:400]
Dec 16 13:41:40 ema mpp[4228]: mpp_enc: mode vbr bps [69000:1104000:1173000] fps fix [23/1] -> fix [23/1] gop i [46] v [0]
Dec 16 13:41:40 ema mpp[4228]: rc: using rc impl default
[... cut ...]
Dec 16 13:41:40 ema mpp[4228]: mpi_enc_test: chn 0 encoded frame 38   size 77747   qp 51 bps_rt 9299864
Dec 16 13:41:40 ema mpp[4228]: mpi_enc_test: chn 0 found last frame. feof 1
Dec 16 13:41:40 ema mpp[4228]: mpi_enc_test: chn 0 encoded frame 39   size 0      
Dec 16 13:41:40 ema mpp[4228]: mpi_enc_test: chn 0 found last packet
Dec 16 13:41:40 ema mpp[4228]: mpi_enc_test: chn 0 encode 40 frames time 161 ms delay   2 ms fps 247.37 bps 15491604
Dec 16 13:41:40 ema mpp[4228]: mpi_enc_test: mpi_enc_test average frame rate 247.37

HermanChen, 2024-12-16:
这编码是正常的，驱动也正常，对比下代码，看上层流程是哪里有问题

suyian7, 2024-12-16:
> 这编码是正常的，驱动也正常，对比下代码，看上层流程是哪里有问题

这个生成的encode.h264文件在本的ffplay之后是乱码

HermanChen, 2024-12-16:
input file name: /home/ema/work/rkmpp_demo/rkmpp_encoder/video/oceans.mp4
 
 mpi_enc_test 的输入文件是 YUV 文件，不能是 mp4