rockchip-linux/mpp #743: mpi_dec_test解码avs2得到的yuv图像颜色不对

lhuang1972, 2024-12-17:
mpi_dec_test -t 16777223 -n 50 -i avs2.raw  -o out.yuv得到的yuv图像颜色有色差，-f 参数貌似也没有什么作用

Hanln-77, 2024-12-19:
找到解决了嘛大哥

HermanChen, 2024-12-19:
非 jpeg 解码器是不带格式转换功能的，需要用 rga 来做格式转换

lhuang1972, 2024-12-19:
rga_test提示不支持yuv420sp10le，且rga_fmt_map也没有yuv420sp10le格式
static RgaFormat rga_fmt_map(MppFrameFormat fmt)
{
    RgaFormat ret;

    switch (fmt) {
    case MPP_FMT_YUV420P:
        ret = RGA_FMT_YCbCr_420_P;
        break;
    case MPP_FMT_YUV420SP:
        ret = RGA_FMT_YCbCr_420_SP;
        break;
    case MPP_FMT_YUV422P:
        ret = RGA_FMT_YCbCr_422_P;
        break;
    case MPP_FMT_YUV422SP:
        ret = RGA_FMT_YCrCb_422_SP;
        break;
    case MPP_FMT_RGB565:
        ret = RGA_FMT_RGB_565;
        break;
    case MPP_FMT_RGB888:
        ret = RGA_FMT_RGB_888;
        break;
    case MPP_FMT_ARGB8888:
        ret = RGA_FMT_RGBA_8888;
        break;
    default:
        ret = RGA_FMT_BUTT;
        mpp_err("unsupport mpp fmt %d found\n", fmt);
        break;
    }

    return ret;
}

HermanChen, 2024-12-19:
不要用 mpp 里的 rga，这个代码太老了，找一下 github 上的 librga

lhuang1972, 2024-12-19:
github上好多版本的librga，有哪个是官方版本吗？

Choise-ieee, 2024-12-20:
https://github.com/airockchip/librga/tree/main 这个好像是官方的

> github上好多版本的librga，有哪个是官方版本吗？

lhuang1972, 2024-12-20:
> https://github.com/airockchip/librga/tree/main 这个好像是官方的
> 
这个版本你试验成功了吗？我试了一下转完是花屏，不知道是不是哪里弄错了

Choise-ieee, 2024-12-21:
> > https://github.com/airockchip/librga/tree/main 这个好像是官方的
> 
> 这个版本你试验成功了吗？我试了一下转完是花屏，不知道是不是哪里弄错了

是的 我也是同样问题 我也在librga那边发issue了

nyanmisaka, 2024-12-23:
https://github.com/airockchip/librga/issues/100

MPP dump 出来的 MPP_FMT_YUV420SP_10BIT / NV15 已经不是原格式了，布局已经变了，也不包含硬件所需的对齐了，RGA 肯定是读不对的。 `[rawvideo @ 0xaaaadaf35330] Invalid buffer size, packet size 9331200 < expected frame_size 15552000`

https://github.com/rockchip-linux/mpp/blob/331ee68fdde99ca85b8a387b0db863f10235425f/utils/utils.c#L143-L167

包含硬件对齐的第一帧数据是这样，YUVView 没法读 [3840x2160_nv15.zip](https://github.com/user-attachments/files/18228146/3840x2160_nv15.zip)

Choise-ieee, 2024-12-30:
> [airockchip/librga#100](https://github.com/airockchip/librga/issues/100)
> 
> MPP dump 出来的 MPP_FMT_YUV420SP_10BIT / NV15 已经不是原格式了，布局已经变了，也不包含硬件所需的对齐了，RGA 肯定是读不对的。 `[rawvideo @ 0xaaaadaf35330] Invalid buffer size, packet size 9331200 < expected frame_size 15552000`
> 
> https://github.com/rockchip-linux/mpp/blob/331ee68fdde99ca85b8a387b0db863f10235425f/utils/utils.c#L143-L167
> 
> 包含硬件对齐的第一帧数据是这样，YUVView 没法读 [3840x2160_nv15.zip](https://github.com/user-attachments/files/18228146/3840x2160_nv15.zip)

那如果想转成nv12或者其他YUV格式 有相关的guide么 谢谢

nyanmisaka, 2024-12-30:
1. 不要dump，直接用rga从mpp解出的fd中读
2. 没有，stride/pitch需要填对

Choise-ieee, 2025-01-06:
> 1. 不要dump，直接用rga从mpp解出的fd中读
> 2. 没有，stride/pitch需要填对

谢谢大佬 ok了 谢谢