rockchip-linux/mpp #744: 实例mpi_dec_mt_test咨询

Choise-ieee, 2024-12-17:
您好
我想咨询下，目前我把MPP最新版本的mpi_dec_mt_test做到codeblocks开发工程中，但是我发现我用codeblocks的程序去运行他就会提示segmentation falut，然后log会提示说Add encoder internal buffer sync，我这个是一个decode的程序为啥提示我要encoder呢
但是我对这个编译好的程序，直接./运行是ok的，这个是什么原因会导致呢
两个打印输出分别为：
Dec 18 06:49:33 lubancat mpp[615515]: mpi_dec_utils: input file /home/cat/MPP/AVS2CCTV-4K.ts_512.dat size 159777627
Dec 18 06:49:33 lubancat mpp[615515]: mpi_dec_utils: cmd parse result:
Dec 18 06:49:33 lubancat mpp[615515]: mpi_dec_utils: input  file name: /home/cat/MPP/AVS2CCTV-4K.ts_512.dat
Dec 18 06:49:33 lubancat mpp[615515]: mpi_dec_utils: output file name: 
Dec 18 06:49:33 lubancat mpp[615515]: mpi_dec_utils: width      :    0
Dec 18 06:49:33 lubancat mpp[615515]: mpi_dec_utils: height     :    0
Dec 18 06:49:33 lubancat mpp[615515]: mpi_dec_utils: type       : 16777223
Dec 18 06:49:33 lubancat mpp[615515]: mpi_dec_utils: max frames :   60
Dec 18 06:49:33 lubancat mpp[615515]: mpp_log: mpi_dec_mt_test start
Dec 18 06:49:33 lubancat mpp[615515]: mpp_log: mpi_dec_mt_test decoder test start w 3840 h 2160 type 16777223
Dec 18 06:49:33 lubancat mpp[615515]: mpp_info: mpp version: ed377c99 author: Herman Chen   2023-12-14 fix[hal_enc]: Add encoder internal buffer sync

Dec 18 06:56:00 lubancat mpp[13109]: mpi_dec_utils: input  file name: /home/cat/MPP/AVS2CCTV-4K.ts_512.dat
Dec 18 06:56:00 lubancat mpp[13109]: mpi_dec_utils: output file name: 
Dec 18 06:56:00 lubancat mpp[13109]: mpi_dec_utils: width      :    0
Dec 18 06:56:00 lubancat mpp[13109]: mpi_dec_utils: height     :    0
Dec 18 06:56:00 lubancat mpp[13109]: mpi_dec_utils: type       : 16777223
Dec 18 06:56:00 lubancat mpp[13109]: mpi_dec_utils: max frames :   60
Dec 18 06:56:00 lubancat mpp[13109]: mpp_log: mpi_dec_mt_test start
Dec 18 06:56:00 lubancat mpp[13109]: mpp_log: mpi_dec_mt_test decoder test start w 3840 h 2160 type 16777223
Dec 18 06:56:00 lubancat mpp[13109]: mpp_info: mpp version: unknown mpp version for missing VCS info
Dec 18 06:56:00 lubancat mpp[13109]: mpp_log: put packet thread start
Dec 18 06:56:00 lubancat mpp[13109]: mpp_log: get frame thread start
Dec 18 06:56:00 lubancat mpp[13109]: mpp_log: decode_get_frame get info changed found
Dec 18 06:56:00 lubancat mpp[13109]: mpp_log: decoder require buffer w:h [3840:2160] stride [4800:2176] size 20889600
Dec 18 06:56:00 lubancat mpp[13109]: mpp_log: 0x55c0c7ef60 decode get frame 0
Dec 18 06:56:03 lubancat mpp[13109]: mpp_log: 0x55c0c7ef60 decode get frame 1
Dec 18 06:56:05 lubancat mpp[13109]: mpp_log: 0x55c0c7ef60 decode get frame 2
Dec 18 06:56:07 lubancat mpp[13109]: mpp_log: 0x55c0c7ef60 decode get frame 3

此外，还想咨询下，4K的avs2的节目解码完了的yuv，对应ffmpeg的格式是哪种呢，尝试用了包括FFMPEG的yuvp10等格式都有些异常，这个是yuv设置为YUV42010bitLE，3840*2160设置的yuv图
[image]

谢谢

HermanChen, 2024-12-19:
Add encoder internal buffer sync  这个是版本信息的 log，看上面一行
解码出来 nv12 的格式，也就是 YUV420SP

lhuang1972, 2024-12-19:
nv12是8bit，好像应该是10bit，YUV420SP10le

HermanChen, 2024-12-19:
这个 avs2 码流是 10bit 的么？10bit 的话，像素数据是紧拼凑排列的

HermanChen, 2024-12-19:
看这个图，亮度部分是对的了，色度部分有问题

Choise-ieee, 2024-12-19:
这个avs2确实是10bit的  而且就是标准的节目流 是的 感觉色度部分偏了 我这边也做了很多尝试 发现还是有问题
sdk里面的这个dump_mpp_frame_to_file写的是ok的
所以就很puzzled

Choise-ieee, 2024-12-19:
我从TS剥离到ES的应该也是oK的
想请您帮我看看，确实是找不到好的idea了，这个是ES包，谢谢
[final.ts_512.zip](https://github.com/user-attachments/files/18193989/final.ts_512.zip)

lhuang1972, 2024-12-19:
你解出来的yuv应该是对的，只是你的yuv播放器选的像素格式不对，或者这个播放器不支持。uv应该是sp格式的，不是p格式的。

Choise-ieee, 2024-12-19:
我用了ffplay.exe -i G:\1.yuv -pixel_format p010 -video_size 3840x2160

Choise-ieee, 2024-12-19:
> 我用了ffplay.exe -i G:\1.yuv -pixel_format p010 -video_size 3840x2160
我看了定义是
#define AV_PIX_FMT_P010       AV_PIX_FMT_NE(P010BE,  P010LE)
AV_PIX_FMT_P010LE, ///< like NV12, with 10bpp per component, data in the high bits, zeros in the low bits, little-endian
AV_PIX_FMT_P010BE, ///< like NV12, with 10bpp per component, data in the high bits, zeros in the low bits, big-endian
感觉应该是这个，但是还是不大正确，满屏绿色的，请问有好的yuv播放器么，谢谢，求赐教

lhuang1972, 2024-12-19:
试试这个https://github.com/IENT/YUView
yuv format选择custom，勾选uv interleaved

Choise-ieee, 2024-12-20:
> 试试这个https://github.com/IENT/YUView yuv format选择custom，勾选uv interleaved

谢谢，谢谢，已经正常
我来看看用librga怎么缩放到其他格式，比如nv12这种，太感谢了

closed 2024-12-20 by Choise-ieee