rockchip-linux/mpp #747: mpi_dec_test输入为.h264解码生成的yuv文件使用ffmpeg打开之后画面是感觉上面盖了一层导致糊糊的

Hanln-77, 2024-12-19:
1.调用指令解码生成yuv
sudo ./mpi_dec_test -i video.h264 -t 7 -n 148 -o outvideo_mpi_dec_test.yuv 
2.使用ffplay指令播放yuv视频
ffplay -f rawvideo -pixel_format yuv420p -video_size 1080x2280 outvideo_mpi_dec_test.yuv
3.播放mpi的demo解码视频效果图片
[image]
4.使用ffmpeg解码生成yuv
ffmpeg -i video.264 -f rawvideo -pix_fmt yuv420p -y outputvideo_ffmpeg_dec_test.yuv
5.再次使用ffplay指令播放yuv视频
ffplay -f rawvideo -pixel_format yuv420p -video_size 1080x2280 outvideo_ffmpeg_dec_test.yuv
6.播放ffmpeg播放解码的yuv文件
[image]

HermanChen, 2024-12-19:
mpp 解码出来的格式是 yuv420sp

Hanln-77, 2024-12-19:
请问对输入的h264的视频格式有什么要求吗
因为我有在程序中处理p格式和sp格式的色彩转换，我对ffmpeg生成的yuv420p文件进行p处理是正常画面显示在我的屏幕上，进行sp处理画面是正常的只是颜色的深度偏绿色但是整体不会变。
但是这个mpi解码生成出来的在p播放和sp处理的播放颜色也是有变化的，但是mpi解码出来的画面上是完全变了多了好几条横线，不仅仅是色彩。

HermanChen, 2024-12-20:
把图分成亮度和色度部分分开观察下